raspiblitz/raspiblitz #2487: Reckless update fails due to non-English $LANG setting

cricktor, 2021-08-23:
I stumbled over an issue when my reckless update attempt to Bitcoin Core 0.21.1 with Raspiblitz v1.7.0 failed to execute. From the menu dialogs you have no chance to see what happened and why it failed (not the issue here now).

The following lines assume gpg will respond in English, but fails to detect a correct signature if $LANG is set to e.g. de_DE.UTF-8 (which is the case on my Raspiblitz, not sure if that was detected automatically or I was asked at the very beginning of the Raspiblitz setup if I want German language for the OS).
https://github.com/rootzoll/raspiblitz/blob/9c7923dc508e39ebd72b7b914d573eaa16b864ed/home.admin/config.scripts/bitcoin.update.sh#L173
https://github.com/rootzoll/raspiblitz/blob/9c7923dc508e39ebd72b7b914d573eaa16b864ed/home.admin/config.scripts/bitcoin.update.sh#L175

A quick fix might be to set LANG to en_EN.UTF-8 before invoking commands whose response is checked in English only.

openoms, 2021-08-23:
The language is the to US English during the SDcard build:
https://github.com/rootzoll/raspiblitz/blob/9c7923dc508e39ebd72b7b914d573eaa16b864ed/build_sdcard.sh#L237

so yours must have been a custom setting.
Reafactoring the scripts should be kept in mind for in case the RaspiBlitz gets a non-english version.

cricktor, 2021-08-24:
Can't remember when an why I changed this. I'm fine with $LANG to have en_US.UTF-8, only want to have my preferred German keyboard layout to be active. I was a bit puzzled why such a setting breaks some/the script. But making the scripts language agnostic is some pain. I will pay more attention on next update. Thx and keep up the good work!

closed 2021-09-07 by rootzoll