raspiblitz/raspiblitz #2489: Curious: reason for multiple Tor processes?

xanoni, 2021-08-23:
I noticed that Raspiblitz creates a second Tor process for `lnd` via `tor-instance-create`. I'm curious, what is the rationale behind this? If you use Tor in `ClientOnly` mode, shouldn't the stream isolation functionality built into Tor be sufficient? There is [a link in the code comments](https://www.torservers.net/wiki/setup/server#multiple_tor_processes), but it seems dead.

See here: https://github.com/rootzoll/raspiblitz/blob/9c7923dc508e39ebd72b7b914d573eaa16b864ed/home.admin/config.scripts/internet.tor.sh#L100

I've only seen this used before when a server was running both a Relay Node / Exit Node _AND_ at least one Hidden Services at the same time. Tor warns the user if this setup is chosen and explains that it could become a privacy (and probably also a security) risk that can be reduced by creating separate instances. (I guess the problem is that it would be too easy to link the traffic if the process crashes / reaches `AccountingMax` etc.)

openoms, 2021-08-23:
The reason is this issue: https://github.com/lightningnetwork/lnd/issues/5166

but https://github.com/rootzoll/raspiblitz/pull/2419 has the changes to not have multiple instances, rather reload Tor with:
`sudo pkill -sighup tor`
instead of systemctl

openoms, 2021-08-23:
Closing as answered.

closed 2021-08-23 by openoms

xanoni, 2021-08-23:
@openoms 
So let me make sure I understand ... so the problem was that Tor couldn't be restarted without also restarting `lnd`.

Why is that different if you restart a separate LND Tor instance instead of restarting the main instance?

By the way, I think you can also just do `systemctl reload tor`.

openoms, 2021-08-23:
We don't restart tor@lnd, only tor@default now. 
`systemctl` reload or `restart` has similar effect afaik.

xanoni, 2021-08-23:
Ohhh now it all makes sense :) Thank you!

openoms, 2021-08-23:
Turns out that 
`sudo systemctl reload tor`
has the same effect as
`sudo pkill -sighup tor`
so could use either instead of
`sudo systemctl restart tor`:
https://github.com/rootzoll/raspiblitz/issues/1604#issuecomment-904196849