raspiblitz/raspiblitz #2490: [Discussion] [CLN] automatic LN node management with CLBOSS

openoms, 2021-08-23:
As described in the CLBOSS plugin will be available for C-ligthning
https://github.com/rootzoll/raspiblitz/issues/2295#issuecomment-896229124

CLBOSS only requires to deposit funds to the onchain wallet of C-lightning.
The recommended amount to start is ~ 10 million satoshis (0.1 BTC).

It does automatically:
* generate outbound capacity - opens channels
* generate inbound capacity - submarine swaps through the boltz.exchange API
* aware of onchain fees and mempool through c-lightning and makes transactions when fees are low
* manages rebalancing - performs probing
* closes bad channels (inactive or low traffic) - this function needs to activated manually

Overall it is a tool which makes users able to send and receive lightning payments with minimal interaction, basically setting up a routing node by itself.

Details: 
https://github.com/ZmnSCPxj/clboss#operating
https://github.com/ZmnSCPxj/clboss/releases

I think it is a great option especially when compared to LND's autopilot (which is only capable of opening channels), but warnings need to be put out about that the doings of CLBOSS can cost money and running it requires a fair amount of trust in the (fully open-source - MIT) code.
Neither the CLBOSS nor the RaspiBlitz developers can take resposibility for lost sats, use at your own discretion!

In any case would like to make it available to be activated as a #reckless, but seemingly very comfortable option. Opinions and experiences will be very welcome!

willcl-ark, 2021-08-23:
FWIW I have been running this for ~6 months on a Debian 10 machine + C-Lightning with no issues.

It opened 6 channels for me which have then worked flawlessly for my (small) payments ever since, requiring no maintenance on my side.

Current in/out balance is:

```text
avail_out=0.0475... btc
avail_in=0.0332... btc
```

...and it is doing rebalancing etc. whenever it thinks it is necessary. I have not tried any extremely large payments, but at that stage it's more up to C-Lightning whether it will succeed.

I installed the plugin to `/usr/local/bin` and then flag it as an `important-plugin` in c-lightning config using:

```
important-plugin=/usr/local/bin/clboss
```

...which means that c-lightning won't startup if it can't load the plugin; might not be what everyone wants, but I wanted to make sure it was always loaded.

I have just (today) activated the new experimental clboss feature `clboss-auto-close=true`, so will keep an eye out to see if it closes any channels it deems as "useless" in the near future.

All-in-all, I haven't had any issues with the plugin and think it might definitely be of interest to someone who just wants to setup a new lightning node, drop a UTXO in there and from then on have the node manage liquidity and channels on its own. 👍🏼

openoms, 2021-08-23:
Thank you for sharing the experiences Will!

I will list some nodeID-s I am told running CLBOSS:
This one has been using it since ~ October 2020:
https://ln.fiatjaf.com/node/0253029d39ca97c4d271af8926b4f3fec37a3aa87b5e538329845f2fc76146fa28
It's activity seems reasonable: 9 channels opened, 9 opened, no scary chain fees (except the period when the mempool was very busy).

openoms, 2021-09-20:
CLBOSS will be available in the SETTINGS menu 
https://github.com/rootzoll/raspiblitz/commit/36e25dc3c61604234d360f9e5e5854cb73cf2c10
[image]

steepdawn974, 2021-10-11:
Hi there. So, I have CLBOSS enabled in the SETTINGS menu on 1.7.1. rc3 .. but I'm not sure it's actually running. 

```
admin@raspberrypi:~ $ lightning-cli clboss-status
{
   "code": -32601,
   "message": "Unknown command 'clboss-status'"
}
```

I would expect this to show up in the lightnind command line as `--plugin=/home/bitcoin/cl-plugins-enabled/clboss` but it doesnt- What am I doing wrong? 

```admin@raspberrypi:~ $ ps -ef | grep lightningd
bitcoin   3131     1  0 13:05 ?        00:00:00 /bin/sh -c /usr/local/bin/lightningd --conf=/home/bitcoin/.lightning/signet/config
bitcoin   3132  3131  0 13:05 ?        00:00:01 /usr/local/bin/lightningd --conf=/home/bitcoin/.lightning/signet/config
bitcoin   3246     1  0 13:05 ?        00:00:00 /bin/sh -c /usr/local/bin/lightningd --conf=/home/bitcoin/.lightning/testnet/config
bitcoin   3248  3246  0 13:05 ?        00:00:03 /usr/local/bin/lightningd --conf=/home/bitcoin/.lightning/testnet/config
bitcoin  27985     1  0 16:03 ?        00:00:00 /bin/sh -c (cat /dev/shm/.cl.pw;echo;cat /dev/shm/.cl.pw) | /usr/local/bin/lightningd --conf=/home/bitcoin/.lightning/config --encrypted-hsm
bitcoin  27989 27985  0 16:03 ?        00:00:24 /usr/local/bin/lightningd --conf=/home/bitcoin/.lightning/config --encrypted-hsm```