raspiblitz/raspiblitz #2491: [UPDATE] manual script downloads for broken updates

openoms, 2021-08-23:
Because of changes in the configuration handling some of the services cannot be updated to the latest any more.
Here I list the updated scripts which can be used now on v1.7 as well.

The https://github.com/rootzoll/raspiblitz/blob/dev/CHANGES.md from the dev branch will have the proposed updates listed.

# RTL
```
# download in place from the dev branch
wget https://raw.githubusercontent.com/rootzoll/raspiblitz/dev/home.admin/config.scripts/bonus.rtl.sh -O config.scripts/bonus.rtl.sh
# remove
config.scripts/bonus.rtl.sh off
# install RTL v0.11.0
config.scripts/bonus.rtl.sh on
```

# Specter Desktop
```
# download in place from the dev branch
wget https://raw.githubusercontent.com/rootzoll/raspiblitz/dev/home.admin/config.scripts/bonus.specter.sh -O config.scripts/bonus.specter.sh
# remove old (keep the data)
config.scripts/bonus.cryptoadvance-specter.sh off
# install Specter Desktop v1.5.1 (be aware the script name has changed so the menu entry will not work)
config.scripts/bonus.specter.sh on
```

rootzoll, 2021-08-23:
Just an idea: Maybe projects could put a special file in their repo that signals "BREAKING_CHANGES.md" - so that RaspiBlitz update function could see beforehand that the new version on GitHub would not work?

openoms, 2021-08-23:
That would be good practice, but unfortunately these changes weren't obvious to me even from the release notes.

As a workaround in the UPDATE me we might offer updates to specific versions, not just latest.

rootzoll, 2021-08-24:
But that could only be UPDATEs to "specific versions" ... not "downgrades".

openoms, 2021-08-25:
> But that could only be UPDATEs to "specific versions" ... not "downgrades".

is there a reason to not allow downgrades? 
Making the script parse the last releases would make it possible to choose any version from the list including going back to a previous one.

rootzoll, 2021-09-09:
Every project that has some kind of database (persisted state) might have problems with "downgrades" - dev normally perpare for upgrading data, but not downgrading. Allowing downgrades can get everything into a mess quickly.

openoms, 2021-09-09:
Agreed. I was thinking about the case when for example there is a project which is installed at version v0.5 by default.
v0.6 comes out and updates fine.
lets say v0.7 breaks.
Currently there would be no choice to choose v0.6 only to reinstall the default v0.5 or update to the latest which is broken.

By downgrading I should have wrote:
allow installing a version other than the latest.

rootzoll, 2021-09-11:
I mean downgrading can be at your own risk ... if it messes things up ... it messes things up.

openoms, 2021-09-12:
Instead of asking repeatedly in the support chat we can link to a github issue where the discussion is less ephemeral and can be updated by anyone testing a new version.

ruethxx, 2021-09-12:
Hi,
sorry, I don't know if I am allowed to type in here... but I am trying to figure out why I can not install/turn on RTL.
Using the script mentioned above I get this (as I used to before):

admin@raspberrypi:~ $ wget https://raw.githubusercontent.com/rootzoll/raspiblitz/dev/home.admin/config.scripts/bonus.rtl.sh -O config.scripts/bonus.rtl.sh
--2021-09-12 22:59:12--  https://raw.githubusercontent.com/rootzoll/raspiblitz/dev/home.admin/config.scripts/bonus.rtl.sh
Resolving raw.githubusercontent.com (raw.githubusercontent.com)... 185.199.111.133, 185.199.108.133, 185.199.110.133, ...
Connecting to raw.githubusercontent.com (raw.githubusercontent.com)|185.199.111.133|:443... connected.
HTTP request sent, awaiting response... 200 OK
Length: 19124 (19K) [text/plain]
Saving to: ‘config.scripts/bonus.rtl.sh’

config.scripts/bonu 100%[===================>]  18.68K  --.-KB/s    in 0.008s  

2021-09-12 22:59:12 (2.29 MB/s) - ‘config.scripts/bonus.rtl.sh’ saved [19124/19124]

admin@raspberrypi:~ $ # remove
admin@raspberrypi:~ $ config.scripts/bonus.rtl.sh off
# Running: 'bonus.rtl.sh off'
config.scripts/bonus.rtl.sh: line 22: /home/admin/config.scripts/network.aliases.sh: No such file or directory
# LNTYPE()
# CHAIN()
# netprefix()
# portprefix()
# typeprefix()
# RTLHTTP()
# configEntry(rtlWebinterface)
# systemdService(RTL)
# missing parameter
admin@raspberrypi:~ $ # install RTL v0.11.0
admin@raspberrypi:~ $ config.scripts/bonus.rtl.sh on
# Running: 'bonus.rtl.sh on'
config.scripts/bonus.rtl.sh: line 22: /home/admin/config.scripts/network.aliases.sh: No such file or directory
# LNTYPE()
# CHAIN()
# netprefix()
# portprefix()
# typeprefix()
# RTLHTTP()
# configEntry(rtlWebinterface)
# systemdService(RTL)
# missing parameter
admin@raspberrypi:~ $ 

So it appears to be installing and then something happens, line 22, no file or directory?! Sorry, I am really an amature -.-

Well, maybe somebody has an idea.
Anyways, thank you for doing this great project. :)

openoms, 2021-09-12:
@ruethxx sorry, that won't work now since other scripts are needed from the dev branch too.

for now just reinstall RTL from the service menu and the new version will be available with the new v1.7.1 SDcard release coming soon.

ruethxx, 2021-09-13:
hey :) thank you for the reply. allright, so reinstall... do you really mean re-install or un-install? :D sorry - confusion :) because it ain't installing 👎  
Yeah, I've never updated the Blitz, looking forward to it. :) you have a release date yet? Sorry, this is no Q&A, right? :D 
Thank you :)

closed 2023-11-14 by rootzoll