raspiblitz/raspiblitz #2492: How to configure that the display power off after 10 minutes and power on on touch?

cniweb, 2021-08-23:
I would like to configure that the display is going to sleep when nobody using the system.
One Wakeup event is when anybody uses the touchscreen.
Is there a preferred way?

openoms, 2021-08-23:
There is no way to switch the screen off as it is lacking the physical switch to do that. 
Could program it to display only a black screen at best.
For programming look at the Blitz TUI.
Here is some documentation to get started: https://github.com/rootzoll/raspiblitz/blob/v1.7/home.admin/BlitzTUI/docs/tutorial.md

NilByte, 2021-08-24:
Something like this might work.

https://github.com/timothyhollabaugh/pi-touchscreen-timeout

rootzoll, 2021-08-24:
>Something like this might work.
>https://github.com/timothyhollabaugh/pi-touchscreen-timeout

@fusion44 had you any luck so far controling the LCD backlight? Maybe this script can help?

openoms, 2021-08-24:
> Something like this might work.
> 
> https://github.com/timothyhollabaugh/pi-touchscreen-timeout

This is for the "Official Raspberry Pi touchscreen", not the one we are using.

There is no way to control the backlight of this 3.5" screen without modifying the PCB.

cniweb, 2021-08-24:
Is this possible (over GPIO 18):
https://www.waveshare.com/wiki/3.5inch_RPi_LCD_(C)#Use_GPIO_to_control_backlight_brightness

ChuckNorrison, 2021-08-28:
you dont need to solder the pcb, just add a switch into the cable. not tested yet, but i would like to modify this behaviour too

openoms, 2021-08-28:
I remember seeing a video where someone soldered a simple switch on the back of the PCB to toggle the power supply of the screen, but can't find it now.