raspiblitz/raspiblitz #2495: Copying from another HDD on the same Pi

byroncoetsee, 2021-08-24:
I see in the FAQ how to copy existing blockchain data from another device on the network. As part of that documentation, it says "But we don't copy the data via USB to the device"...
What if I want to? I've got the other HDD plugged into the Pi - how do I copy it across?

rootzoll, 2021-08-24:
Call the script "sudo /home/admin/XXcopyStation.sh" .. its a script to copy from one HDD to many other HDDs connected on a powerfull USB-Hub.

byroncoetsee, 2021-08-24:
Hey thanks for the response.
I just tried running it and after "copying bitcoin to template folder" (or something of the sort), it started the loop over and found my external drive. It then wanted to start formatting it? Is that expected?

I've got a hdd attached to the pi (primary.. synced to about 3% or so)
I've also got an ssd attached (secondary.. synced to 100%)

I'd rather not have the ssd formatted. haha

byroncoetsee, 2021-08-24:
Ah hang on - looking through the script code I think I may have misunderstood things...
It looks like that script is used to copy from an external hdd which had raspiblitz (It currently has Umbrel - I was hoping to be able to specify the dir manually).

I'll try do the copy manually. Thanks for the help!

closed 2021-08-24 by rootzoll