raspiblitz/raspiblitz #2498: Running JM yield generator fills /tmp

m00ninite, 2021-08-25:
If the yield generator is running, the raspiblitz status screen will dump the joinmarket maker info into /tmp every few seconds. Eventually this fills up the SD card, causing the device to crash (a month or 2 of uptime). The blitz can only be recovered if you pop it into another computer and nuke /rootfs/tmp.

There are thousands of files like this:
[image]

Bug appears to be in [home.admin/00infoBlitz.sh](https://github.com/rootzoll/raspiblitz/blob/fc1d44104bc22d85ebebd7b0d6ee2240722a9edf/home.admin/00infoBlitz.sh#L289)

Proposed solution is to clear the previous temp file when writing a new one. or just reusing the same one. This does bring up another question though.. should /tmp be mounted as a RAM disk rather than to the SD card? Writing tmp files to the SD can cause a lot of thrashing on an already fragile storage medium.

openoms, 2021-08-31:
Thanks, should create the tempfile in `/dev/shm` as in the rest of the scripts.

rootzoll, 2021-09-16:
@openoms whats the status on this for v1.7.1?

openoms, 2021-09-17:
Changed in most places focusing on the important ones like passwords or the ones building up (running often). Can be closed.

closed 2021-09-20 by rootzoll