raspiblitz/raspiblitz #2506: Error Update Mempool

GabrielRCL, 2021-08-28:
I'm not able to update the mempool, could someone help me?
[image]

i use git -c but nothing happens
[image]

i change cp to mv in:
https://github.com/rootzoll/raspiblitz/blob/79ccb8d3433a59ee8268b458436dbb4f3a824aa3/home.admin/config.scripts/bonus.mempool.sh#L358
but not working too

nyxnor, 2021-08-28:
```bash
sudo cp /home/mempool/mempool/mempool-config.json /home/admin
pinnedVersion="v2.1.2"
sudo -u mempool git -C /home/mempool/mempool reset --hard $pinnedVersion
bash config.scripts/bonus.mempool.sh update
```

GabrielRCL, 2021-08-31:
> Try this
> 
> ```shell
> sudo cp /home/mempool/mempool/mempool-config.json /home/admin
> pinnedVersion="v2.1.2"
> sudo -u mempool git -C /home/mempool/mempool reset --hard $pinnedVersion
> bash config.scripts/bonus.mempool.sh update
> ```

the directory "/home/mempool/mempool/" was missing the file "mempool-config.json". however, I found it in the directory "/home/admin".
I made a copy using the inverse command:
"sudo cp /home/admin/mempool-config.json /home/mempool/mempool"

but hasn't updated yet, although he says he's updated.

[image]

closed 2021-09-01 by GabrielRCL

nyxnor, 2021-09-13:
leave this open, it is not fixed yet. also permission ddenied is because of the user you are running on

rootzoll, 2021-09-13:
v1.7.1 will come with the latest mempool v2.2.2

GabrielRCL, 2021-09-13:
> leave this open, it is not fixed yet. also permission ddenied is because of the user you are running on

it seems that it was solved yes, it returned some errors but it worked.