raspiblitz/raspiblitz #2507: Wallet recovery fail with AttributeError: 'bytes' object has no attribute 'encode'

lessless, 2021-08-29:
Hi,

I'm trying to run a recovery script to restore a wallet:  `lnd.initwallet.py scb mypassword  “twenty four words of seeds“ 2021-08-29-channel.backup`

but it throws an error: 

```
# OK SCB file exists
# *** RECOVERING LND WALLET FROM SEED + SCB ***
b'65...ab'
Traceback (most recent call last):
  File "/home/lnd/bin/lnd.initwallet.py", line 281, in <module>
    main()
  File "/home/lnd/bin/lnd.initwallet.py", line 273, in main
    scb(stub, wallet_password, seed_words, seed_password, file_path_scb)
  File "/home/lnd/bin/lnd.initwallet.py", line 120, in scb
    channel_backups=scb_hex_str.encode()
AttributeError: 'bytes' object has no attribute 'encode'
```

lessless, 2021-08-29:
cc @frennkie (due to https://github.com/rootzoll/raspiblitz/pull/841/files#diff-be8867ca998cffd1217366230c977acda82fdf7c693d6d47615d7faa16173af7R123)

openoms, 2021-08-31:
The script is not designed to be run outside of the the raspiblitz environment.

Follow the https://github.com/lightningnetwork/lnd/blob/master/docs/recovery.md#recovering-using-scbs on any system running LND:

`lncli create -multi_file=channel.backup`

closed 2021-09-09 by rootzoll