raspiblitz/raspiblitz #2508: FEATURE REQUEST: Initial sync choice TESTNET vs MAINNET

we-do-it-lu, 2021-08-29:
When someone installs Raspiblitz for the first time the time required by the download of the complete MAINNET is an unnecessary barrier all he wants is play with his Blitz ASAP, would be great to have the option "sync TESTNET" vs "sync MAINNET" at first start.

rootzoll, 2021-08-30:
Because we the recent refactoring of the setup options for the v1.7.1 we decided to not have the "TESTNET" as an option from the beginning and offer Testnet & Signet as an option you can run in parallel once MAINNET is running.

I know that makes it not easy&fast to setup .. but making sure that the whole setup process is working now with also c-lightning coming as an option with v1.7.1 would double all the testing efforts. Keep in mind that also with v1.8 we plan to introduce another optional setup process with the WebUI next to the SSH - fo first we need to get that stable.

Once the new setup processes are implemented and well tested we might take a look how to provide a quick testnet setup without mainnet .. but I will push this up to v1.8.1 to take another look.

we-do-it-lu, 2021-09-07:
Thank you @rootzoll, makes sense.

closed 2024-07-08 by rootzoll