raspiblitz/raspiblitz #2510: Cannot fund LNbits

benjamin-wilson, 2021-09-03:
After installing LNbits, I cannot fund a wallet because "self-payments not allowed". I don't see a way to fund the lnbits wallet without a second node.

rootzoll, 2021-09-09:
Yeah ... my old workaround is to install a mobile wallet - send there and then send back into LNbits.

@arcbtc is there a easier way in the meanwhile?

openoms, 2021-11-08:
This is an LNBits limitation for now. Keep an eye on their decelopments. Closing.

ZeroPointEnergy, 2022-11-24:
Just stumbled over this problem. It's possible without a second wallet by looping a payment with the CLI `lncli payinvoice --allow_self_payment <invoice>`

rootzoll, 2022-11-26:
I marked this for a later released to check what options developed in the meantime.

closed 2023-11-14 by rootzoll