raspiblitz/raspiblitz #2513: Electrum Rust Server running but Electrum wallet is "not connected"

Nerfzooka, 2021-09-05:
Issue #1895 is happening again.  
Raspiblitz v 1.7.0 running 3 services:  BTCPayServer, BTC-RPC-Explorer and Electrum Rust Server  can explore blockchain and transactions but my Electrum wallet on the same network says"not connected".
Tried rebooting, reindexing, restarting, no change.  Double checked that the Electrum wallet automatic server selection is off and the rust ip and port are right.  Nginx is running the BTC-RPC and BTCPayServer websites fine. I can ping the raspiblitz from the wallet computer and ping the wallet computer from the raspiblitz.  Here's a weird observation:  the Electrum wallet block count is 2 higher than the raspiblitz.
[image]

rootzoll, 2021-09-21:
Can you post the status report you get when in the RaspiBlitz menu you go `ELECTRS` and then `STATUS`?

Nerfzooka, 2021-09-22:
Yep
[image]
.

rootzoll, 2021-10-10:
Next week the v1.7.1 will get released with an updated Electrs ... please try with that updated version and report in if the problem still persists.

Nerfzooka, 2021-10-17:
Updated to 1.7.1 and have a slightly different problem.  Electrum wallet still says "not connected", but BTC-RPC Explorer is also not working now with a "502 Bad Gateway".  On the bright side the BTCPay server is serving its web page, so nginx must be working.

[image]

Nerfzooka, 2021-10-18:
Hmm... Is electrs still indexing?

[image]

Nerfzooka, 2021-10-22:
Hey, Electrum wallet is connected after I rebooted the raspiblitz to remove lcd screen .  👍

Nerfzooka, 2021-10-22:
Electrum wallet connected to another server, not mine.  I thought it was fixed for a minute, but it's not.

Nerfzooka, 2021-11-01:
Ah, I need electrum command line to delete certificates.  Is there a way to do that on windows?

openoms, 2021-11-03:
@Nerfzooka  just empty the certs from User/appdata/roaming/electrum/certs/

Nerfzooka, 2021-11-04:
Hey @openoms , I'm so happy, Clearing out the cert folder fixed the problem.   Yay!

lupn3, 2022-10-05:
Raspiblitz 1.8.0 running, full synced

Electrs
- is activated
- has opened port 50002
- is connected to localhost

but suddenly out of the nowhere the connection to the wallet isn't possible anymore. It seems that Electrs is receiving the request from the wallet (not: electrum wallet), but not responding to it, even after reinstall an re-indexing Electrs.

Nerfzooka, 2022-10-20:
Oh no, @lupn3, after updating Raspiblitz to 1.8.0 I've got the same problem again too.  
Electrs is running, All is synced, All is indexed. BTC-RPC-Explorer is running normally, but electrum wallet is not connected.  The wallet syncs immediately if I choose another server, but shows no connection when looking at the raspiblitz.  Emptying the electrum wallets cert folder didn't work this time.
[image]

lupn3, 2022-10-21:
@Nerfzooka Have opened an additional issue here (https://github.com/rootzoll/raspiblitz/issues/3343). Unfortunately, I still have not made any progress. Superficially everything looks good, but the Raspi still refuses any connection to the wallet. It is exasperating.

Nerfzooka, 2022-10-21:
TLDR: This was fixed for me by uninstalling then reinstalling electrs on the Raspiblitz and deleting electrum wallet certificates from the desktop.

closed 2022-10-21 by Nerfzooka