raspiblitz/raspiblitz #2516: Risk of not being able to booth with btrfs after kernel update

ephestione, 2021-09-06:
I appreciate how this flavor of node installation considers btrfs as a filesystem for the LN data, albeit with experimental support.

What I have read around, and this is a good example:
https://feldspaten.org/2019/07/05/raspbian-and-btrfs/
is that Raspberry OS will not automatically update the initramfs after a kernel update, so a partition running on btrfs won't be able to boot unless you remember, each time a kernel update occurs, to manually run mkinitramfs *before* rebooting.

I searched for "mkinitramfs" and"initramfs" in the code base and found nothing, and "kernel" returns results that are not relevant to this issue.

Have you tested a kernel update on the system when btrfs is used, or is there no need to account for initramfs in a 64bit system?

rootzoll, 2021-09-21:
Marking this issue for later .. after v1.8.0 I want to take a look at BRTFS and booting options again.

rootzoll, 2024-11-14:
Closing this issue because we consider to remove BTRFS from RaspiBlitz ... see: https://github.com/raspiblitz/raspiblitz/issues/2045#issuecomment-2471845122

closed 2024-11-14 by rootzoll