raspiblitz/raspiblitz #2520: Premature end build_sdcard.sh: "!!! BUILD FAILED --> PGP Verify not OK / signature(0) verify(0)"

nxtester, 2021-09-07:
Hi,
running build_sdcard.sh breaks always with message:   "!!! BUILD FAILED --> PGP Verify not OK / signature(0) verify(0)"

Output:
----------------------------------------------

*** PREPARING BITCOIN ***
--2021-09-07 18:47:10--  https://bitcoin.org/laanwj-releases.asc
Auflösen des Hostnamens bitcoin.org (bitcoin.org)… 172.67.40.154, 2606:4700:10::6816:44b0, 2606:4700:10::ac43:289a, ...
Verbindungsaufbau zu bitcoin.org (bitcoin.org)|172.67.40.154|:443 … verbunden.
HTTP-Anforderung gesendet, auf Antwort wird gewartet … 200 OK
Länge: 32242 (31K) [application/octet-stream]
Wird in »laanwj-releases.asc« gespeichert.

laanwj-releases.asc             100%[======================================================>]  31,49K  --.-KB/s    in 0,06s

2021-09-07 18:47:10 (544 KB/s) - »laanwj-releases.asc« gespeichert [32242/32242]

gpg: key 90C8019E36C2E964: 51 Beglaubigungen wegen fehlender Schlüssel nicht geprüft
pub   rsa4096 2015-06-24 [SC] [verfällt: 2022-02-10]
      01EA5486DE18A882D4C2684590C8019E36C2E964
uid                      Wladimir J. van der Laan (Bitcoin Core binary release signing key) <laanwj@gmail.com>

gpg: key 90C8019E36C2E964: 51 Beglaubigungen wegen fehlender Schlüssel nicht geprüft
gpg: Schlüssel 90C8019E36C2E964: Öffentlicher Schlüssel "Wladimir J. van der Laan (Bitcoin Core binary release signing key) <laanwj@gmail.com>" importiert
gpg: Anzahl insgesamt bearbeiteter Schlüssel: 1
gpg:                              importiert: 1
gpg: keine ultimativ vertrauenswürdigen Schlüssel gefunden
--2021-09-07 18:47:10--  https://bitcoin.org/bin/bitcoin-core-0.21.0/SHA256SUMS.asc
Auflösen des Hostnamens bitcoin.org (bitcoin.org)… 172.67.40.154, 2606:4700:10::ac43:289a, 2606:4700:10::6816:45b0, ...
Verbindungsaufbau zu bitcoin.org (bitcoin.org)|172.67.40.154|:443 … verbunden.
HTTP-Anforderung gesendet, auf Antwort wird gewartet … 200 OK
Länge: 1769 (1,7K) [application/octet-stream]
Wird in »SHA256SUMS.asc« gespeichert.

SHA256SUMS.asc                  100%[======================================================>]   1,73K  --.-KB/s    in 0s

2021-09-07 18:47:12 (23,3 MB/s) - »SHA256SUMS.asc« gespeichert [1769/1769]

goodSignature(0)
correctKey(0)

!!! BUILD FAILED --> PGP Verify not OK / signature(0) verify(0)

-----------------------------------------------------------------------------------

Any ideas? Thank you!

Best regards

rootzoll, 2021-09-09:
Hi ... please try the build script of the dev branch (v1.7.1rc1) - thats the update we are working on with all the updated PGP checks. https://github.com/rootzoll/raspiblitz/blob/dev/build_sdcard.sh

nxtester, 2021-09-10:
Hi, thanks for your support. 
Tried it, but problem still exists:

OUTPUT
------------------------------------------------------------------------------------------

*** PREPARING BITCOIN ***
--2021-09-10 18:56:44--  https://bitcoin.org/laanwj-releases.asc
Auflösen des Hostnamens bitcoin.org (bitcoin.org)… 104.22.69.176, 2606:4700:10::6816:45b0, 2606:4700:10::ac43:289a, ...
Verbindungsaufbau zu bitcoin.org (bitcoin.org)|104.22.69.176|:443 … verbunden.
HTTP-Anforderung gesendet, auf Antwort wird gewartet … 200 OK
Länge: 32242 (31K) [application/octet-stream]
Wird in »laanwj-releases.asc« gespeichert.

laanwj-releases.asc                 100%[==================================================================>]  31,49K  --.-KB/s    in 0,04s

2021-09-10 18:56:44 (779 KB/s) - »laanwj-releases.asc« gespeichert [32242/32242]

gpg: key 90C8019E36C2E964: 51 Beglaubigungen wegen fehlender Schlüssel nicht geprüft
pub   rsa4096 2015-06-24 [SC] [verfällt: 2022-02-10]
      01EA5486DE18A882D4C2684590C8019E36C2E964
uid                      Wladimir J. van der Laan (Bitcoin Core binary release signing key) <laanwj@gmail.com>

gpg: key 90C8019E36C2E964: 51 Beglaubigungen wegen fehlender Schlüssel nicht geprüft
gpg: Schlüssel 90C8019E36C2E964: Öffentlicher Schlüssel "Wladimir J. van der Laan (Bitcoin Core binary release signing key) <laanwj@gmail.com>" importiert
gpg: Anzahl insgesamt bearbeiteter Schlüssel: 1
gpg:                              importiert: 1
gpg: keine ultimativ vertrauenswürdigen Schlüssel gefunden
--2021-09-10 18:56:44--  https://bitcoin.org/bin/bitcoin-core-0.21.1/SHA256SUMS.asc
Auflösen des Hostnamens bitcoin.org (bitcoin.org)… 104.22.69.176, 2606:4700:10::ac43:289a, 2606:4700:10::6816:44b0, ...
Verbindungsaufbau zu bitcoin.org (bitcoin.org)|104.22.69.176|:443 … verbunden.
HTTP-Anforderung gesendet, auf Antwort wird gewartet … 200 OK
Länge: 1778 (1,7K) [application/octet-stream]
Wird in »SHA256SUMS.asc« gespeichert.

SHA256SUMS.asc                      100%[==================================================================>]   1,74K  --.-KB/s    in 0s

2021-09-10 18:56:45 (36,0 MB/s) - »SHA256SUMS.asc« gespeichert [1778/1778]

goodSignature(0)
correctKey(0)

!!! BUILD FAILED --> PGP Verify not OK / signature(0) verify(0)
root@rb001:/home/rpi#

rootzoll, 2021-09-21:
OpenOms merged a fix for UK users ... can you try the build script from `dev` branch again?
https://github.com/rootzoll/raspiblitz/blob/dev/build_sdcard.sh

rootzoll, 2021-12-16:
The install of bitcoin, LND & c-lightning is now called in build script just by install scripts. Will be part of v1.7.2 release candidates. Closing issue.

closed 2021-12-16 by rootzoll