raspiblitz/raspiblitz #2523: BTC-RPC-Explorer (v3.2.0) says "no address server available" with default configuration

david-bakin, 2021-09-09:
This isn't necessarily a solid bug report, but this is what I observed:

Raspiblitz 1.7.0 + Etherrs _upgraded to 0.8.11 manually_ + BTC-RCP-Explorer _upgraded to 3.2.0 manually_.  (Both those upgrades are currently in the post 1.7.0 dev branch.)

As configured that way, hitting BTC-RPC-Explorer with 

```
http://<username>:<password>@192.168.86.250:3020/address/3QMduh6zdz1n7LFpq7AwJa3LR7zWtSwdKr
```

gives a page with _no transactions_ (should have 2), instead it says "no address server configured" and points you to the sample `.env` in its repo.

However, I manually changed `/home/btcrpcexplorer/.config/btc-rpc-explorer.env` from

```
BTCEXP_ELECTRUMX_SERVERS=tcp://127.0.0.1:50001
```

to

```
BTCEXP_ELECTRUM_SERVERS=tcp://127.0.0.1:50001
```

(that is `_ELECTRUMX` to `_ELECTRUM`) and now it works.  Yes, I know that just one line above that it says _both_ work, but this is my observation.

(BTW, that address isn't important: it's just some random address I plucked off of `blockchain.info` today.)

rootzoll, 2021-09-16:
I have the BTC-RCP-Explorer 3.2.0 & electes  0.8.11 running in my test build of v1.7.1RC2 and it works with the `BTCEXP_ELECTRUMX_SERVERS`.

I will set this for v1.7.1 on final testing, but if there are no further error reports on the upcomming release candidates, than this should work fine.

rootzoll, 2021-10-10:
found some small bug in the install .. retest on a rc3 with patched dev code on a fresh device needed - plus then testing on recover.

rootzoll, 2021-10-11:
OK test ran good. Closing issue for release.

closed 2021-10-11 by rootzoll