raspiblitz/raspiblitz #2525: Connection reset by ip port 22

elidauchiko, 2021-09-10:
HI there,
  I have just received my new raspiblitz v 1.7 and I cannot ssh into the node.
I always get this message in the terminal: Connection reset by IP port 22. I have pinged it and it connects. So I am stuck. Can anyone help? I dearly want to get my node up and running.

rootzoll, 2021-09-11:
Have you tried to flash a fresh image of RaspiBlitz v1.7.0 to the sd card?
https://github.com/rootzoll/raspiblitz#downloading-the-software

elidauchiko, 2021-09-11:
Thanks for the advice. I will try this. Can you point me to some very detailed instructions on how to flash the SD card foe someone who is computer illiterate? Thanks.

rootzoll, 2021-09-11:
see the instructions here: https://github.com/rootzoll/raspiblitz#write-the-sd-card-image-to-your-sd-card ... the Balena Ether is a quite easy app to flash an image to an sd card

elidauchiko, 2021-09-11:
Thanks for you advice but because I am such a computer illiterate I can't seem to download balena etcher onto my Linux elementary. Maybe there is another app I can use. Anyway I don't want to use you your valuable time for such a Luddite like me.

rootzoll, 2021-09-12:
Check if you can find Etcher of some other SD card tools on your Linux Distros App Store.

With most Linux Distros you should even be able to write an image to an sd card without any extra app. Download the image file (its an img.gz file) from right-click menu unzip/unpack the file to a (img file) and then often on right-click menu there is the option to write image to a drive - there you choose the sd card and then its flashing.

If your Linux distro is does not have the last option .. you can also go to the terminal and use the `dd` command to write an image to a drive. But thats just your last option.

elidauchiko, 2021-09-15:
Dear rootzoll, I have managed to download balena etcher and the software but now when I use balena etcher, I have no target to select. What does this mean?

elidauchiko, 2021-09-15:
Thanks everyone I have resolved this issue with help over in telegram . Thanks!

closed 2021-09-15 by rootzoll