raspiblitz/raspiblitz #2526: Stuck on "waiting for chain sync" after multiple reboots.

leemcarthur, 2021-09-11:
Hi there. My node has been stuck on "waiting for chain sync" after multiple reboots. I suspect this has to do with a failed installation of thunderhub, but I could use some assistance with troubleshooting. Any help would be appreciated!

My log output is here: https://termbin.com/q1uj

openoms, 2021-09-11:
wait until `progress=1` in the bitcoin debug.log

leemcarthur, 2021-09-11:
[image]

leemcarthur, 2021-09-11:
My current thinking is that it is related to me turning on "Run Behind TOR". Other than enabling that setting and letting it reboot itself following turning that setting on, am I supposed to do something else to make this work properly behind Tor?

leemcarthur, 2021-09-11:
Fresh logs: https://termbin.com/1jame

openoms, 2021-09-11:
Please use the latest image v1.7.

openoms, 2021-11-07:
Here the script is only detecting if:
`lncli getinfo`
shows:
`synced_to_chain": true`

Please check your LND logs for errors:
for example with:
`sudo tail -n 30 -f /mnt/hdd/lnd/logs/bitcoin/mainnet/lnd.log`
or 
`sudo journalctl -fu lnd`

ge0rg0, 2021-11-08:
This is the message I get in the lnd.log. When a host is unreachable, it gets stuck on Waiting for Chain Sync...

2021-11-08 10:49:54.503 [DBG] BTCN: Failed to connect to 02844ed44f730789d50af924fe96b81d59e123d77f11883a9a0d7e40b3038a8824@sebzays2m4altv2w3lx22lgxdl7grcsm4wj2cigwkcxotk4dvle5ktqd.onion:9735 (reqid 5): socks connect tcp 127.0.0.1:9050->sebzays2m4altv2w3lx22lgxdl7grcsm4wj2cigwkcxotk4dvle5ktqd.onion:9735: unknown error host unreachable
2021-11-08 10:49:54.503 [DBG] BTCN: Retrying connection to 02844ed44f730789d50af924fe96b81d59e123d77f11883a9a0d7e40b3038a8824@sebzays2m4altv2w3lx22lgxdl7grcsm4wj2cigwkcxotk4dvle5ktqd.onion:9735 (reqid 5) in 3m10s

openoms, 2021-11-08:
Just to clarify is this just about the display being stuck or there is a problem with LND (or Tor) as well?

openoms, 2021-11-08:
Indeed, seems that the display script is working as intended.

What does 
`lncli getinfo`
show?

any errors in the lnd or bitcoin logs?

ge0rg0, 2021-11-08:
I don't have any errors either on lnd or bitcoin logs
I checked my lnd.conf and I'm not using neutrino

```
[Bitcoin]
bitcoin.active=1
bitcoin.node=bitcoind
# enable either testnet or mainnet
#bitcoin.testnet=1
bitcoin.mainnet=1

```

Private info editet:

```
lncli getinfo
{
    "version": "0.13.3-beta commit=v0.13.3-beta",
    "commit_hash": "*********************************************************************",
    "identity_pubkey": "******************************************************",
    "alias": "******",
    "color": "#68f442",
    "num_pending_channels": 0,
    "num_active_channels": 12,
    "num_inactive_channels": 1,
    "num_peers": 13,
    "block_height": 708783,
    "block_hash": "*******************************************************",
    "best_header_timestamp": "1636373821",
    "synced_to_chain": false,
    "synced_to_graph": true,
    "testnet": false,
    "chains": [
        {
            "chain": "bitcoin",
            "network": "mainnet"
        }
    ],
    "uris": [
        "<node onion address>"
    ],
    "features": {
        "0": {
            "name": "data-loss-protect",
            "is_required": true,
            "is_known": true
        },
        "5": {
            "name": "upfront-shutdown-script",
            "is_required": false,
            "is_known": true
        },
        "7": {
            "name": "gossip-queries",
            "is_required": false,
            "is_known": true
        },
        "9": {
            "name": "tlv-onion",
            "is_required": false,
            "is_known": true
        },
        "12": {
            "name": "static-remote-key",
            "is_required": true,
            "is_known": true
        },
        "14": {
            "name": "payment-addr",
            "is_required": true,
            "is_known": true
        },
        "17": {
            "name": "multi-path-payments",
            "is_required": false,
            "is_known": true
        },
        "23": {
            "name": "anchors-zero-fee-htlc-tx",
            "is_required": false,
            "is_known": true
        },
        "30": {
            "name": "amp",
            "is_required": true,
            "is_known": true
        },
        "31": {
            "name": "amp",
            "is_required": false,
            "is_known": true
        }
    }
}

```

openoms, 2021-11-08:
the problem is :
`"synced_to_chain": false,`

cstenglein, 2023-06-07:
@leemcarthur  is this still an issue with 1.9.0?

Closing this for now, please re-open if it still persists.

closed 2023-06-07 by cstenglein