raspiblitz/raspiblitz #2528: this is just a question....

DrBlackross, 2021-09-11:
is there a way to migrate this to a pc (or rack mount) x86 install version of raspiblitz? or is there a x86 version of raspiblitz to migrate to?

my retro pi NES i think is cramping my node lol

openoms, 2021-09-11:
No there is no supported way to do this yet. Keep an eye on https://github.com/rootzoll/raspiblitz/issues/1500, will update soon with instruction to run in an Ubuntu 20.04 VM.

closed 2021-09-12 by openoms

DrBlackross, 2021-09-13:
k, btw the NES from retropi was running great till the fan bearings inside had other plans. And i have a few 1ru cases but i  don't want to move the RPI4 into that, i'd rather just migrate to a x86 system and use the LND backup gzip to restore everything.

with post #1500  will it allow the original 1tb drive from the rpi setup? and the backup...... (i still don't quite get docker images, kubernetes, yes... docker eh)

either way i'll follow #1500

DrBlackross, 2021-09-13:
(wife just got pissed i just covered her side of the bed with the 1ru and an old a68hm with a a6-7400k, wires and few ssd's)

gonna try #1500

DrBlackross, 2021-09-15:
@openoms, my take on #1500 .....

ok 
figured out the riddle here

------------------------------------------------------------------------------------------------------------
https://github.com/rootzoll/raspiblitz/issues/1386

Not tested or supported, but worth a try. Will likely need some custom work. Be ready to debug and modify some scripts.
Here are the rough principles:
https://github.com/rootzoll/raspiblitz/tree/master/alternative.platformsare
.....
Flash A Compatible Image To The Usb Drive (Not A Disk - Need To Use An Equivalent To The Sdcard On The Rpi).
So Set Up Debian Or Ubuntu To A Usb. Boot And Run The Build_Sdcard.Sh.
.....
The scripts will format the internal SSD for the blockchain and data.
From there you will need to figure out.
It helps if you run an RPI4 node so you know how it should work.
------------------------------------------------------------------------------------------------------------

made net install usb drive of ubuntu (mini.iso to flash drive)

i needed a raid0 setup because my original 1tb ssd is in my rpi4 node which is running... i am trying to shift it over to a AMD x86 system, so down to the junk pile i go.
i find and old amd a6-7400k (with 8gb ram), a 32gbssd, 956gb ssd (which rapsiblitz won't except unless its 1tb solid or more, damit), and a 434gb ssd.
installed base system of ubuntu 20.04 with ssh and basic server install optoins then rebooted..... done

git cloned raspiblitz on default user i created when installing 20.04

tried to get the 2 ssds to raid0..... ugh, 15 yrs its been and i hate mdadm, well mdadm changed over the years and is easy. nope....
i get the raid0 setup on every try (i've repeated install of os, and raspiblitz x7 now)
mdadm keeps wiping after rebooting and installing raspi blitz, or changing it from md0 to md127, etc...

then found this

https://github.com/rootzoll/raspiblitz/issues/1683

__________________________________________________________________
(this next part is for RAID0 array setup

BTRFS

sure..... works.
https://linuxhint.com/install-and-use-btrfs-on-ubuntu-lts/

make sure you set the raid mount to /mnt/hdd

Ex. sudo mkfs.btrfs -f -m raid0 -L /mnt/hdd /dev/sdb1 /dev/sdc1 
(after x2 full os and raspiblitz reinstalls, total x9)

so when you add it to the fstab it mounts to /mnt/hdd
__________________________________________________________________

EITHER WAY, make sure whatever SSD/HDD your using automounts to /mnt/hdd, raspiblitz will move forward and partition and format that "blockchain" drive /mnt/hdd

git cloned raspiblitz on default user i used when installing 20.04 (again)

run ./build_sdcard.sh

**********************************************
BASIC SD CARD BUILD DONE			<--------------- always says good!
**********************************************

Your SD Card Image for RaspiBlitz is ready (might still do display config).
Take the chance & look thru the output above if you can spot any errors or warnings.

IMPORTANT IF WANT TO MAKE A RELEASE IMAGE FROM THIS BUILD:	<----------- this part is where i get the blue screen on reboot also it doesnt see the raid drives
1. login fresh --> user:admin password:raspiblitz
2. run --> ./XXprepareRelease.sh

REBOOT

login with default login to check the raid, for me worked!

ok with BTRFS!

open another ssh terminal this time to admin@thatip, usually i have seizures from the flashing blue screen
but this time it worked, choose freshsetup and then it will skip to the hdd creation and not do it..... 

(also don't forget to have whatever 1tb drive your going to use automount as /mnt/hdd)

what you have to make shure to have btrfs raid drives setup if using multiple drives (it will drop to a BTRFS menu and setup partitions and then work)

Ex. sudo mkfs.btrfs -f -m raid0 -L /mnt/hdd /dev/sdb1 /dev/sdc1 

next youll get to the bitcoin litecoin etc screen, i chose litecoin (for testing) BUT.... itll install the arm binaries, and try to run litecoind (which won't start), 
in which case... just go compile litecoind for x86 and make install then run.
-----
used this site for btc and/or ltc
(db-4.8.30 is a pain in 20.04)
https://docs.hydrachain.org/guide-for-linux
-----

make; make install for either litecoin or bitcoin

reboot

ssh admin@thatip

  Blockchain Progress :  0.00 % 0 peers  
  Lightning Progress  :  21.71 % 0 peers
  Please wait - this can take some time  
  Use CTRL+c to EXIT to Terminal

seems to have worked

run 'menu'

installing stuff just to see if it installs and runs (bos, rtl, etc)

RTL up
BOS up
sudo su - 					
MemPool up

and remove default user account i was using to work in background, also disable suspend mode (scared myself thought i broke it when i deleted default account.

sudo systemctl mask sleep.target suspend.target hibernate.target hybrid-sleep.target
verify
sudo systemctl status sleep.target suspend.target hibernate.target hybrid-sleep.target

good was worried....

everything is running, lightcoind, and lnd.... just waiting for it to sync up

DrBlackross, 2021-09-15:
in build_sdcard.sh
btrfs_tools needs to be btrfs-progs

DrBlackross, 2021-09-15:
btrfs still works great!

BUT i simplified it a bit

admin:~/raspiblitz$ nano /home/admin/raspiblitz.info
baseimage=ubuntu
cpu=x86_64
displayClass=headless

*****************************************
* RASPIBLITZ SD CARD IMAGE SETUP v1.7.1 *
*****************************************
For details on optional parameters - see build script source code:
1) will use NO-INTERACTION --> 'false'
2) will use FATPACK --> 'false'
3) will use GITHUB-USERNAME --> 'rootzoll'
4) will use GITHUB-BRANCH --> 'v1.7'
5) will use DISPLAY-CLASS --> 'headless'
6) will use TWEAK-BOOTDRIVE --> 'false'
7) will use WIFI --> 'false'
X) will use CPU-ARCHITECTURE --> 'x86_64'
X) will use OPERATINGSYSTEM ---> 'ubuntu'
Do you agree with all parameters above? (yes/no) yes
Building RaspiBlitz ...

(BTW)
*** raspiblitz.info *** 
baseimage=ubuntu
cpu=x86_64
displayClass=headless

**********************************************
BASIC SD CARD BUILD DONE
**********************************************

Your SD Card Image for RaspiBlitz is ready (might still do display config).
Take the chance & look thru the output above if you can spot any errors or warnings.

IMPORTANT IF WANT TO MAKE A RELEASE IMAGE FROM THIS BUILD:
1. login fresh --> user:admin password:raspiblitz
2. run --> release

# BUILD DONE - see above

reboot

edit these lines

admin:~/raspiblitz$ nano config.scripts/blitz.litecoin.sh
  # set version (change if update is available)
  litecoinVersion="0.18.1"
  litecoinSHA256="ca50936299e2c5a66b954c266dcaaeef9e91b2f5307069b9894048acf3eb5751"

  # cleaning download folder 
  sudo rm -r /home/admin/download 1>/dev/null
  sudo -u admin mkdir -p /home/admin/download
  cd /home/admin/download

  # download
  binaryName="litecoin-${litecoinVersion}-x86_64-linux-gnu.tar.gz"
  sudo -u admin wget https://download.litecoin.org/litecoin-${litecoinVersion}/linux/${binaryName} 1>/dev/null
  
should work

DrBlackross, 2021-09-15:
OS: Ubuntu 20.04.3 LTS x86_64 
Host: MS-7721 9.0 
Kernel: 5.4.0-84-generic 
Uptime: 16 mins 
Packages: 908 (dpkg) 
Shell: bash 5.0.17 
Resolution: 4096x2160 
Terminal: /dev/pts/0 
CPU: AMD A6-7400K Radeon R5 2C+4G (2) @ 3.500GHz 
GPU: AMD ATI Radeon R5 Graphics 
Memory: 1501MiB / 6978MiB