raspiblitz/raspiblitz #2531: [UbuntuVM] permission issues of tor@default

openoms, 2021-09-12:
Using the current state of the `dev` branch I am running into an annoying issue with the `tor@default` instance. Interestingly `tor@lnd` works flawlessly.

Settings are at default and tried already reinstalling and deleting the `/mnt/hdd/tor` directory to no avail.

```
$ tor --version
Tor version 0.4.5.10.
$ uname -a
Linux nadmin-BHYVE 5.11.0-34-generic #36~20.04.1-Ubuntu SMP Fri Aug 27 08:06:32 UTC 2021 x86_64 x86_64 x86_64 GNU/Linux
```

The failure is:

```
$ sudo journalctl -n 20 -fu tor@default
-- Logs begin at Sat 2021-09-11 07:47:51 BST. --
Sep 12 13:46:54 nadmin-BHYVE tor[407859]: Sep 12 13:46:54.020 [notice] Tor can't help you if you use it wrong! Learn how to be safe at https://www.torproject.org/download/download#warning
Sep 12 13:46:54 nadmin-BHYVE tor[407859]: Sep 12 13:46:54.020 [notice] Read configuration file "/usr/share/tor/tor-service-defaults-torrc".
Sep 12 13:46:54 nadmin-BHYVE tor[407859]: Sep 12 13:46:54.020 [notice] Read configuration file "/etc/tor/torrc".
Sep 12 13:46:54 nadmin-BHYVE tor[407859]: Sep 12 13:46:54.022 [notice] Opening Socks listener on 127.0.0.1:9050
Sep 12 13:46:54 nadmin-BHYVE tor[407859]: Sep 12 13:46:54.022 [notice] Opened Socks listener connection (ready) on 127.0.0.1:9050
Sep 12 13:46:54 nadmin-BHYVE tor[407859]: Sep 12 13:46:54.022 [notice] Opening Control listener on 127.0.0.1:9051
Sep 12 13:46:54 nadmin-BHYVE tor[407859]: Sep 12 13:46:54.022 [notice] Opened Control listener connection (ready) on 127.0.0.1:9051
Sep 12 13:46:54 nadmin-BHYVE tor[407859]: Sep 12 13:46:54.022 [warn] Directory /mnt/hdd/tor/sys cannot be read: Permission denied
Sep 12 13:46:54 nadmin-BHYVE tor[407859]: Sep 12 13:46:54.022 [notice] Closing partially-constructed Socks listener connection (ready) on 127.0.0.1:9050
Sep 12 13:46:54 nadmin-BHYVE tor[407859]: Sep 12 13:46:54.022 [notice] Closing partially-constructed Control listener connection (ready) on 127.0.0.1:9051
Sep 12 13:46:54 nadmin-BHYVE tor[407859]: Sep 12 13:46:54.022 [warn] Failed to parse/validate config: Couldn't create private data directory "/mnt/hdd/tor/sys"
Sep 12 13:46:54 nadmin-BHYVE tor[407859]: Sep 12 13:46:54.022 [err] Reading config failed--see warnings above.
Sep 12 13:46:54 nadmin-BHYVE systemd[1]: tor@default.service: Main process exited, code=exited, status=1/FAILURE
Sep 12 13:46:54 nadmin-BHYVE systemd[1]: tor@default.service: Failed with result 'exit-code'.
Sep 12 13:46:54 nadmin-BHYVE systemd[1]: Failed to start Anonymizing overlay network for TCP.
Sep 12 13:46:54 nadmin-BHYVE systemd[1]: tor@default.service: Scheduled restart job, restart counter is at 5.
Sep 12 13:46:54 nadmin-BHYVE systemd[1]: Stopped Anonymizing overlay network for TCP.
Sep 12 13:46:54 nadmin-BHYVE systemd[1]: tor@default.service: Start request repeated too quickly.
Sep 12 13:46:54 nadmin-BHYVE systemd[1]: tor@default.service: Failed with result 'exit-code'.
Sep 12 13:46:54 nadmin-BHYVE systemd[1]: Failed to start Anonymizing overlay network for TCP.
```

openoms, 2021-09-12:
* the service fails
```
$ sudo systemctl status tor@default
● tor@default.service - Anonymizing overlay network for TCP
     Loaded: loaded (/lib/systemd/system/tor@default.service; enabled-runtime; vendor preset: enabled)
    Drop-In: /etc/systemd/system/tor@default.service.d
             └─raspiblitz.conf
     Active: failed (Result: exit-code) since Sun 2021-09-12 18:59:52 BST; 1min 44s ago
    Process: 60245 ExecStartPre=/usr/bin/install -Z -m 02755 -o debian-tor -g debian-tor -d /run/tor (code=exited, status=0/SUCCESS)
    Process: 60246 ExecStartPre=/usr/bin/tor --defaults-torrc /usr/share/tor/tor-service-defaults-torrc -f /etc/tor/torrc --RunAsDaemon 0 --verify-config (code=exited, status=0/SUCCESS)
    Process: 60247 ExecStart=/usr/bin/tor --defaults-torrc /usr/share/tor/tor-service-defaults-torrc -f /etc/tor/torrc --RunAsDaemon 0 (code=exited, status=1/FAILURE)
   Main PID: 60247 (code=exited, status=1/FAILURE)
```
* but this works:
```
sudo -u debian-tor tor -f /etc/tor/torrc --RunAsDaemon 0
```

openoms, 2021-09-12:
SUCCES

disabling the line:
```
AppArmorProfile=-system_tor
```
in `/lib/systemd/system/tor@default.service`
did it!

now Tor is working.

openoms, 2021-09-14:
Reopening as might need a permament solution if reproduced in Ubuntu based setups.

openoms, 2021-09-15:
Could reproduce the same problem and fix on a new install of Ubuntu 20.04 LTS in a VM on TrueNAS.
```
$ uname -a
Linux nadmin-BHYVE 5.11.0-34-generic #36~20.04.1-Ubuntu SMP x86_64 x86_64 x86_64 GNU/Linux
```

For reference the apparmor status (this is not used on the RPi):
```
admin@nadmin-BHYVE:~$ sudo aa-status                                                                     
apparmor module is loaded.                                                                               
38 profiles are loaded.                                                                                  
36 profiles are in enforce mode.                                                                         
   /snap/snapd/12704/usr/lib/snapd/snap-confine                                                          
   /snap/snapd/12704/usr/lib/snapd/snap-confine//mount-namespace-capture-helper                          
   /snap/snapd/12883/usr/lib/snapd/snap-confine                                                          
   /snap/snapd/12883/usr/lib/snapd/snap-confine//mount-namespace-capture-helper                          
   /usr/bin/evince                                                                                       
   /usr/bin/evince-previewer                                                                             
   /usr/bin/evince-previewer//sanitized_helper                                                           
   /usr/bin/evince-thumbnailer                                                                           
   /usr/bin/evince//sanitized_helper                                                                     
   /usr/bin/man                                                                                          
   /usr/lib/NetworkManager/nm-dhcp-client.action                                                         
   /usr/lib/NetworkManager/nm-dhcp-helper                                                                
   /usr/lib/connman/scripts/dhclient-script                                                              
   /usr/lib/cups/backend/cups-pdf                                                                        
   /usr/lib/snapd/snap-confine                                                                           
   /usr/lib/snapd/snap-confine//mount-namespace-capture-helper                                           
   /usr/sbin/cups-browsed
   /usr/sbin/cupsd   
   /usr/sbin/cupsd//third_party
   /usr/sbin/tcpdump         
   /{,usr/}sbin/dhclient
   ippusbxd                                            
   libreoffice-senddoc 
   libreoffice-soffice//gpg
   libreoffice-xpdfimport
   lsb_release        
   man_filter  
   man_groff     
   nvidia_modprobe     
   nvidia_modprobe//kmod
   snap-update-ns.snap-store
   snap.snap-store.hook.configure
   snap.snap-store.snap-store
   snap.snap-store.ubuntu-software
   snap.snap-store.ubuntu-software-local-file
   system_tor     
2 profiles are in complain mode.
   libreoffice-oopslash
   libreoffice-soffice
2 processes have profiles defined.     
2 processes are in enforce mode.                                     
   /usr/sbin/cups-browsed (134616)                     
   /usr/sbin/cupsd (134614)
0 processes are in complain mode.
0 processes are unconfined but have a profile defined.           
```

rootzoll, 2021-11-30:
Merged PR .. setting this on final testing.

openoms, 2021-12-06:
fixed in #2745 closing

openoms, 2022-01-10:
@nyxnor this fix afbc484 has been reverted in `dev` and the problem surfaced again on x86_64 Debian.
Any reason to not implement it again?

nyxnor, 2022-01-10:
the system_tor apparmor config is on `/etc/apparmor.d/system_tor`.
here is what I have in it
```
# vim:syntax=apparmor
#include <tunables/global>

profile system_tor flags=(attach_disconnected) {
  #include <abstractions/tor>

  owner /var/lib/tor/** rwk,
  owner /var/lib/tor/ r,
  owner /var/log/tor/* w,

  # During startup, tor (as root) tries to open various things such as
  # directories via check_private_dir().  Let it.
  /var/lib/tor/** r,

  /{,var/}run/tor/ r,
  /{,var/}run/tor/control w,
  /{,var/}run/tor/socks w,
  /{,var/}run/tor/tor.pid w,
  /{,var/}run/tor/control.authcookie w,
  /{,var/}run/tor/control.authcookie.tmp rw,
  /{,var/}run/systemd/notify w,

  # Site-specific additions and overrides. See local/README for details.
  #include <local/system_tor>
}
```

If we think about security, disabling the apparmor set by debian to use with tor is not the safest idea.
Can we try modifying the apparmor config to include this new directory?
Or we can first analyze the apparmor configuration differences on Debian X Ubuntu regarding system_tor.

openoms, 2022-01-10:
Agree that adapting is better than disabling.
My file on Debian Bullseye X86:
```
$ cat /etc/apparmor.d/system_tor
# vim:syntax=apparmor
#include <tunables/global>

profile system_tor flags=(attach_disconnected) {
  #include <abstractions/tor>

  owner /var/lib/tor/** rwk,
  owner /var/lib/tor/ r,
  owner /var/log/tor/* w,

  # During startup, tor (as root) tries to open various things such as
  # directories via check_private_dir().  Let it.
  /var/lib/tor/** r,

  /{,var/}run/tor/ r,
  /{,var/}run/tor/control w,
  /{,var/}run/tor/socks w,
  /{,var/}run/tor/tor.pid w,
  /{,var/}run/tor/control.authcookie w,
  /{,var/}run/tor/control.authcookie.tmp rw,
  /{,var/}run/systemd/notify w,

  # Site-specific additions and overrides. See local/README for details.
  #include <local/system_tor>
}
```

On Ubuntu it is the same:
```
$ cat /etc/apparmor.d/system_tor
# vim:syntax=apparmor
#include <tunables/global>

profile system_tor flags=(attach_disconnected) {
  #include <abstractions/tor>

  owner /var/lib/tor/** rwk,
  owner /var/lib/tor/ r,
  owner /var/log/tor/* w,

  # During startup, tor (as root) tries to open various things such as
  # directories via check_private_dir().  Let it.
  /var/lib/tor/** r,

  /{,var/}run/tor/ r,
  /{,var/}run/tor/control w,
  /{,var/}run/tor/socks w,
  /{,var/}run/tor/tor.pid w,
  /{,var/}run/tor/control.authcookie w,
  /{,var/}run/tor/control.authcookie.tmp rw,
  /{,var/}run/systemd/notify w,

  # Site-specific additions and overrides. See local/README for details.
  #include <local/system_tor>
}
```

nyxnor, 2022-01-10:
so not a difference... I don't know, chaning the apparmor config is the last resort on ubuntu to mount an outside data directory.

openoms, 2022-01-10:
Ok this apparmor profile with the added directories does work:
```
# vim:syntax=apparmor
#include <tunables/global>

profile system_tor flags=(attach_disconnected) {
  #include <abstractions/tor>

  owner /var/lib/tor/** rwk,
  owner /var/lib/tor/ r,
  owner /var/log/tor/* w,

  owner /mnt/hdd/tor/** rwk,
  owner /mnt/hdd/tor/ r,
  owner /mnt/hdd/app-data/tor/** r,

  # During startup, tor (as root) tries to open various things such as
  # directories via check_private_dir().  Let it.
  /var/lib/tor/** r,
  /mnt/hdd/tor/** r,
  /mnt/hdd/app-data/tor/** r,

  /{,var/}run/tor/ r,
  /{,var/}run/tor/control w,
  /{,var/}run/tor/socks w,
  /{,var/}run/tor/tor.pid w,
  /{,var/}run/tor/control.authcookie w,
  /{,var/}run/tor/control.authcookie.tmp rw,
  /{,var/}run/systemd/notify w,

  # Site-specific additions and overrides. See local/README for details.
  #include <local/system_tor>
}
```

rootzoll, 2022-02-16:
Closing for v1.7.2 release.

PR #2745 (2021-11-30): VM fixes - ongoing testing
fix #2531
PR #2876 (2022-01-10): [x86] correct apparmor.d/system_tor
fix #2531
closed 2022-02-16 by rootzoll