raspiblitz/raspiblitz #2532: Cannot start LND.  Error while dialing dial tcp 127.0.0.1:10009

mdmtranch, 2021-09-12:
After a reboot I'm unable to start LND. 

After unlocking my LND wallet and running `lncli getinfo` I'm shown the following error: 
`[lncli] rpc error: code = Unavailable desc = connection error: desc = "transport: Error while dialing dial tcp 127.0.0.1:10009: connect: connection refused"`

Debug log: 
[https://termbin.com/wut8](https://termbin.com/wut8)

I've tried a fresh install but no luck. Anyone had a similar issue?

openoms, 2021-09-12:
Seem to be some LND errors. Check more with:
`sudo journalctl  -n100 -fu lnd`

mdmtranch, 2021-09-12:
Seems to be some sort of panic error loop. 

`panic: invalid page type: 15615: 10`

```
-- Logs begin at Sun 2021-09-12 14:15:30 BST. --
Sep 12 14:25:23 raspberrypi lnd[6751]: github.com/lightningnetwork/lnd/macaroons.(*Service).UnaryServerInterceptor.func1(0x1118240, 0x40023e10e0, 0xd87e80, 0x40000fff80, 0x40023d5360, 0x4001d65d10, 0x4001ed5500, 0x0, 0x40020fb9c8, 0x8da64c)
Sep 12 14:25:23 raspberrypi lnd[6751]:         github.com/lightningnetwork/lnd/macaroons/service.go:202 +0x150
Sep 12 14:25:23 raspberrypi lnd[6751]: github.com/grpc-ecosystem/go-grpc-middleware.ChainUnaryServer.func1(0x1118240, 0x40023e10e0, 0xd87e80, 0x40000fff80, 0x40023d5360, 0x40023d5380, 0xd5b540, 0x40023e10e0, 0x40023ea600, 0x400049d600)
Sep 12 14:25:23 raspberrypi lnd[6751]:         github.com/grpc-ecosystem/go-grpc-middleware@v1.0.0/chain.go:39 +0x138
Sep 12 14:25:23 raspberrypi lnd[6751]: github.com/lightningnetwork/lnd/lnrpc._Lightning_GetNodeInfo_Handler(0xe274a0, 0x400013a2d0, 0x1118240, 0x40023e10e0, 0x40023ea600, 0x40000b11a0, 0x1118240, 0x40023e10e0, 0x40023e2f00, 0x46)
Sep 12 14:25:23 raspberrypi lnd[6751]:         github.com/lightningnetwork/lnd/lnrpc/rpc.pb.go:16124 +0x128
Sep 12 14:25:23 raspberrypi lnd[6751]: google.golang.org/grpc.(*Server).processUnaryRPC(0x400049d600, 0x11262e0, 0x4001ad2900, 0x4001ed5500, 0x40000b1320, 0x190eb20, 0x0, 0x0, 0x0)
Sep 12 14:25:23 raspberrypi lnd[6751]:         google.golang.org/grpc@v1.24.0/server.go:995 +0x384
Sep 12 14:25:23 raspberrypi lnd[6751]: google.golang.org/grpc.(*Server).handleStream(0x400049d600, 0x11262e0, 0x4001ad2900, 0x4001ed5500, 0x0)
Sep 12 14:25:23 raspberrypi lnd[6751]:         google.golang.org/grpc@v1.24.0/server.go:1275 +0xab0
Sep 12 14:25:23 raspberrypi lnd[6751]: google.golang.org/grpc.(*Server).serveStreams.func1.1(0x4000a1b640, 0x400049d600, 0x11262e0, 0x4001ad2900, 0x4001ed5500)
Sep 12 14:25:23 raspberrypi lnd[6751]:         google.golang.org/grpc@v1.24.0/server.go:710 +0x8c
Sep 12 14:25:23 raspberrypi lnd[6751]: created by google.golang.org/grpc.(*Server).serveStreams.func1
Sep 12 14:25:23 raspberrypi lnd[6751]:         google.golang.org/grpc@v1.24.0/server.go:708 +0x8c
Sep 12 14:25:23 raspberrypi systemd[1]: lnd.service: Main process exited, code=exited, status=2/INVALIDARGUMENT
Sep 12 14:25:23 raspberrypi systemd[1]: lnd.service: Failed with result 'exit-code'.
Sep 12 14:26:23 raspberrypi systemd[1]: lnd.service: Service RestartSec=1min expired, scheduling restart.
Sep 12 14:26:23 raspberrypi systemd[1]: lnd.service: Scheduled restart job, restart counter is at 2.
Sep 12 14:26:23 raspberrypi systemd[1]: Stopped LND Lightning Daemon.
Sep 12 14:26:23 raspberrypi systemd[1]: Starting LND Lightning Daemon...
Sep 12 14:26:23 raspberrypi systemd[1]: Started LND Lightning Daemon.
Sep 12 14:33:00 raspberrypi lnd[12205]: panic: invalid page type: 15615: 10
Sep 12 14:33:00 raspberrypi lnd[12205]: goroutine 1806 [running]:
Sep 12 14:33:00 raspberrypi lnd[12205]: go.etcd.io/bbolt.(*Cursor).search(0x40009c6c68, 0x400179be30, 0x29, 0x29, 0x80dd)
Sep 12 14:33:00 raspberrypi lnd[12205]:         go.etcd.io/bbolt@v1.3.5-0.20200615073812-232d8fc87f50/cursor.go:250 +0x2a0
Sep 12 14:33:00 raspberrypi lnd[12205]: go.etcd.io/bbolt.(*Cursor).searchPage(0x40009c6c68, 0x400179be30, 0x29, 0x29, 0x7f5989a000)
Sep 12 14:33:00 raspberrypi lnd[12205]:         go.etcd.io/bbolt@v1.3.5-0.20200615073812-232d8fc87f50/cursor.go:308 +0x10c
Sep 12 14:33:00 raspberrypi lnd[12205]: go.etcd.io/bbolt.(*Cursor).search(0x40009c6c68, 0x400179be30, 0x29, 0x29, 0x5298)
Sep 12 14:33:00 raspberrypi lnd[12205]:         go.etcd.io/bbolt@v1.3.5-0.20200615073812-232d8fc87f50/cursor.go:265 +0x11c
Sep 12 14:33:00 raspberrypi lnd[12205]: go.etcd.io/bbolt.(*Cursor).searchPage(0x40009c6c68, 0x400179be30, 0x29, 0x29, 0x7f59344000)
Sep 12 14:33:00 raspberrypi lnd[12205]:         go.etcd.io/bbolt@v1.3.5-0.20200615073812-232d8fc87f50/cursor.go:308 +0x10c
Sep 12 14:33:00 raspberrypi lnd[12205]: go.etcd.io/bbolt.(*Cursor).search(0x40009c6c68, 0x400179be30, 0x29, 0x29, 0x4d42)
Sep 12 14:33:00 raspberrypi lnd[12205]:         go.etcd.io/bbolt@v1.3.5-0.20200615073812-232d8fc87f50/cursor.go:265 +0x11c
Sep 12 14:33:00 raspberrypi lnd[12205]: go.etcd.io/bbolt.(*Cursor).searchPage(0x40009c6c68, 0x400179be30, 0x29, 0x29, 0x7f5773a000)
Sep 12 14:33:00 raspberrypi lnd[12205]:         go.etcd.io/bbolt@v1.3.5-0.20200615073812-232d8fc87f50/cursor.go:308 +0x10c
Sep 12 14:33:00 raspberrypi lnd[12205]: go.etcd.io/bbolt.(*Cursor).search(0x40009c6c68, 0x400179be30, 0x29, 0x29, 0x3138)
Sep 12 14:33:00 raspberrypi lnd[12205]:         go.etcd.io/bbolt@v1.3.5-0.20200615073812-232d8fc87f50/cursor.go:265 +0x11c
Sep 12 14:33:00 raspberrypi lnd[12205]: go.etcd.io/bbolt.(*Cursor).seek(0x40009c6c68, 0x400179be30, 0x29, 0x29, 0xa684a2ee97ee5802, 0xacbd859b2b2b8a4d, 0x352d297038ac4b01, 0xb0423d5b86b88d00, 0x1, 0x7f4b2d1e30, ...)
Sep 12 14:33:00 raspberrypi lnd[12205]:         go.etcd.io/bbolt@v1.3.5-0.20200615073812-232d8fc87f50/cursor.go:159 +0x68
Sep 12 14:33:00 raspberrypi lnd[12205]: go.etcd.io/bbolt.(*Bucket).Get(0x4002b7ec00, 0x400179be30, 0x29, 0x29, 0x30, 0xbccbe0, 0x1)
Sep 12 14:33:00 raspberrypi lnd[12205]:         go.etcd.io/bbolt@v1.3.5-0.20200615073812-232d8fc87f50/bucket.go:262 +0x84
Sep 12 14:33:00 raspberrypi lnd[12205]: github.com/btcsuite/btcwallet/walletdb/bdb.(*bucket).Get(0x4002b7ec00, 0x400179be30, 0x29, 0x29, 0x40009c6d58, 0x5bf5a8, 0x40017a1e68)
[... cut ...]
Sep 12 14:33:00 raspberrypi systemd[1]: lnd.service: Failed with result 'exit-code'.
Sep 12 14:34:00 raspberrypi systemd[1]: lnd.service: Service RestartSec=1min expired, scheduling restart.
Sep 12 14:34:00 raspberrypi systemd[1]: lnd.service: Scheduled restart job, restart counter is at 3.
Sep 12 14:34:00 raspberrypi systemd[1]: Stopped LND Lightning Daemon.
Sep 12 14:34:00 raspberrypi systemd[1]: Starting LND Lightning Daemon...
Sep 12 14:34:00 raspberrypi systemd[1]: Started LND Lightning Daemon.

```

openoms, 2021-09-12:
There is an option to easily update LND from the `UPDATE` menu, but it might be related to a channel.db corruption and if funds funds are at risk I would ask in the LND Dev Slack (dev.lightning.community/) or in an LND issue (https://github.com/lightningnetwork/lnd/issues) first.

mdmtranch, 2021-09-17:
> There is an option to easily update LND from the UPDATE menu, but it might be related to a channel.db corruption and if funds are at risk I would ask in the LND Dev Slack (dev.lightning.community/) or in an LND issue (https://github.com/lightningnetwork/lnd/issues) first.
> 
> Also make sure you have a latest channel.backup saved as well.

Updated to the latest version of LND which did not seem to solve the corruption issue. 
Added db.bolt.auto-compact=true to my LND config but seemed not to help. The panic loop still persists.
I've also tried setting up a fresh node and restoring all channels from the latest version of LND to it and still seem to have the same issue with db.bolt.auto-compact=true

Any ideas? 

sudo journalctl -n100 -fu lnd
[https://pastebin.com/GaJDQjL4](https://pastebin.com/GaJDQjL4)

Node Debug log
[https://termbin.com/ukph](https://termbin.com/ukph)

openoms, 2021-09-19:
Please report this to the LND team.
It might a database corruption or an LND bug.

mdmtranch, 2021-09-20:
> It might a database corruption or an LND bug.
Correct, this is a confirmed database corruption issue now. 
[https://github.com/lightningnetwork/lnd/issues/5712](https://github.com/lightningnetwork/lnd/issues/5712)

I'm going to need to try and recover funds. Starting a new node and using `lnd-rescue` seems to have the same db corruption issue. 

Would attempting recovery using the `channel.backup` file and wallet seed enable me to force close all open channels and recover funds?

openoms, 2021-09-20:
> Would attempting recovery using the `channel.backup` file and wallet seed enable me to force close all open channels and recover funds?

yes, as long as your peers are online. Can attempt the recovery repeatedly to catch any offline peers.
See: https://github.com/rootzoll/raspiblitz/blob/v1.7/FAQ.md#2-recover-from-wallet-seed
and 
https://node-recovery.com/

mdmtranch, 2021-09-20:
> yes, as long as your peers are online. Can attempt the recovery repeatedly to catch any offline peers.
> See: https://github.com/rootzoll/raspiblitz/blob/v1.7/FAQ.md#2-recover-from-wallet-seed
> and
> https://node-recovery.com/

Thanks for this. I've begun the process. 
How long does it usually take for LND to finish syncing? 

```
admin@raspberrypi:~ $ sudo tail -n 30 /mnt/hdd/lnd/logs/bitcoin/mainnet/lnd.log
2021-09-20 15:59:30.599 [INF] LNWL: Recovered addresses from blocks 602779-604778
2021-09-20 15:59:40.035 [INF] LNWL: Scanning 2000 blocks for recoverable addresses
2021-09-20 16:10:18.745 [INF] LNWL: Recovered addresses from blocks 604779-606778
2021-09-20 16:10:29.470 [INF] LNWL: Scanning 2000 blocks for recoverable addresses
2021-09-20 16:21:21.481 [INF] LNWL: Recovered addresses from blocks 606779-608778
2021-09-20 16:21:31.767 [INF] LNWL: Scanning 2000 blocks for recoverable addresses
2021-09-20 16:31:03.536 [INF] LNWL: Recovered addresses from blocks 608779-610778
2021-09-20 16:31:14.151 [INF] LNWL: Scanning 2000 blocks for recoverable addresses
2021-09-20 16:41:22.906 [INF] LNWL: Recovered addresses from blocks 610779-612778
2021-09-20 16:41:33.965 [INF] LNWL: Scanning 2000 blocks for recoverable addresses
2021-09-20 16:51:45.168 [INF] LNWL: Recovered addresses from blocks 612779-614778
2021-09-20 16:51:55.795 [INF] LNWL: Scanning 2000 blocks for recoverable addresses
2021-09-20 17:03:37.284 [INF] LNWL: Recovered addresses from blocks 614779-616778
2021-09-20 17:03:48.047 [INF] LNWL: Scanning 2000 blocks for recoverable addresses
2021-09-20 17:15:13.643 [INF] LNWL: Recovered addresses from blocks 616779-618778
2021-09-20 17:15:23.939 [INF] LNWL: Scanning 2000 blocks for recoverable addresses
2021-09-20 17:26:10.932 [INF] LNWL: Recovered addresses from blocks 618779-620778
2021-09-20 17:26:21.221 [INF] LNWL: Scanning 2000 blocks for recoverable addresses
2021-09-20 17:38:47.009 [INF] LNWL: Recovered addresses from blocks 620779-622778
2021-09-20 17:38:57.555 [INF] LNWL: Scanning 2000 blocks for recoverable addresses
2021-09-20 17:49:30.693 [INF] LNWL: Recovered addresses from blocks 622779-624778
2021-09-20 17:49:41.224 [INF] LNWL: Scanning 2000 blocks for recoverable addresses
2021-09-20 18:00:05.776 [INF] LNWL: Recovered addresses from blocks 624779-626778
2021-09-20 18:00:16.078 [INF] LNWL: Scanning 2000 blocks for recoverable addresses
2021-09-20 18:12:26.930 [INF] LNWL: Recovered addresses from blocks 626779-628778
2021-09-20 18:12:37.203 [INF] LNWL: Scanning 2000 blocks for recoverable addresses
2021-09-20 18:25:19.010 [INF] LNWL: Recovered addresses from blocks 628779-630778
2021-09-20 18:25:29.466 [INF] LNWL: Scanning 2000 blocks for recoverable addresses
2021-09-20 18:38:22.879 [INF] LNWL: Recovered addresses from blocks 630779-632778
2021-09-20 18:38:33.334 [INF] LNWL: Scanning 2000 blocks for recoverable addresses
```

cata-code, 2022-02-02:
If you are following the [official install instructions](https://github.com/lightningnetwork/lnd/blob/master/docs/INSTALL.md) then know that you need to have both `bitcoin-core` and `lnd` running, before starting the `lncli create` command