raspiblitz/raspiblitz #2533: Cannot create new IP2TOR subscription for LNBITS

Chris869218, 2021-09-12:
When trying to create a new IP2TOR subscription for LNBITS, the terminal screen displays:

please wait ..
{'timestamp': [TIMESTAMP]} 
None

and then gets stuck there. Tried restarting the RaspiBlitz, but I get the same result. 

Running RaspiBlitz v1.7.0

The background info is my previous static IP from my LNBITS IP2TOR subscription stopped working, so my plan was to cancel the subscription and re-subscribe to see if that fixed the issue. I could connect over TOR, and my local IP, but the IP2TOR IP address was no longer working. Perhaps not being able to re-subscribe is related to why it stopped working in the first place. Any ideas? Thanks!

frennkie, 2021-09-12:
There is an outage on the backend node. Trying to fix it.

frennkie, 2021-09-12:
The backend issue was solved (my SDCARD died and I recovered my RaspiBlitz node). If you still have problems this might be caused by something other/additional. Please let me know.

Chris869218, 2021-09-13:
Unfortunately no change. After selecting "NEW1 + IP2TOR Bridge (paid)" still getting stuck at:

please wait ..
{'timestamp': '2021-09-13T03:46:30Z'}
None

Restarted the RaspiBlitz again just to be safe.

frennkie, 2021-09-14:
We had to setup a new host ("eu-fra2") because the old host (eu-fra1 = 91.109.21.148) had to be shutdown (for unrelated reasons).

Now you should be able to create a new subscription using "eu-fra2". 

If the old IP2Tor subscription is blocking the process you should be able to fix this by hand by going into the config file `/mnt/hdd/app-data/subscriptions/subscriptions.toml`. Just remove the complete section that is related to "eu-fra1".

Chris869218, 2021-09-15:
I had already unsubscribed from my previous subscriptions (confirmed with the /home/admin/config.scripts/blitz.subscriptions.ip2tor.py subscriptions-list command returning [] )

Still getting hung up on the following. Tried rebooting the RaspiBlitz as well just in case that changed anything. 

please wait ..
{'timestamp': '2021-09-15T01:37:55Z'}
None

Any other ideas as to what could be going on? Really appreciate the help.

rootzoll, 2021-09-22:
Tested the IP2Tor for LNbits for the v1.7.1 release. Had to remove DYNU domain service for LetsEncrypt for now, but DuckDNS is still working.

Once v1.7.1 is released - please consider:

- If you run a LetsEncrypt with DYNU consider to change to DuckDNS.
- If your IP2Tor is not working ... cancel and subscribe to one off the new provided in the IP2Tor shop.

closed 2021-09-22 by rootzoll

Chris869218, 2021-09-22:
Thanks, looking forward to v1.7.1