raspiblitz/raspiblitz #2535: Official installation script for Debian x64 platform

nxtester, 2021-09-12:
To push adoption and amount of Lightning full nodes we need beside the Raspberry Pi platform a more universal and tested, official installation script of RaspiBlitz for a recent Debian x64 platform. For example, Debian is widely adopted in virtual server hosting. personal computer x64 platforms and NAS with virtualization platforms like QNAP and Synlogy.

I myself tried also several platforms and feel quite a bit frustrated. Whilst the Raspberry-Pi platform might be ideal for the RaspiBlitz installation, it seems that hardware isn't. Serious bottleneck is limited robustness of any MicroSD memorycard keeping OS and RaspiBlitz software platform. It's simply not designed for such usecase and therefore stability problems are daily routine.

RaspiBlitz and Raspberry-Pi ae great for first steps but not a robust enough 24/7/365 platform.

That's why a approved and official Debian x64 installation script would be great.

openoms, 2021-09-13:
See the work done towards running in an Ubuntu 20.04 VM: https://github.com/rootzoll/raspiblitz/issues/1500#issuecomment-917444439

Debian 11 is unfortunately not running on my TrueNAS based setup.

openoms, 2021-11-30:
There is progress on the build script:
https://github.com/rootzoll/raspiblitz/issues/1500#issuecomment-982606560

The logic is that the scripts would use the second (nonOS) disk for the chain and other storage.

I am running VMs on Ubuntu, but testing on Debian is welcome.

adevoss, 2021-12-22:
Testing on Debian VM #2772

openoms, 2022-01-12:
Successfully installed raspiblitz using the build_script from the dev branch.
1 disk for the OS and used a separate 1TB one for the blockchain.

```
# download the build script
wget https://raw.githubusercontent.com/rootzoll/raspiblitz/dev/build_sdcard.sh
# run
sudo bash build_sdcard.sh false false rootzoll dev hdmi false
```

openoms, 2022-01-12:
with the new changes in the build options it should be:
```
# download the build script
wget -O build_sdcard.sh https://raw.githubusercontent.com/rootzoll/raspiblitz/dev/build_sdcard.sh
# run
sudo bash build_sdcard.sh --github-user rootzoll --branch dev --display hdmi --tweak-boot-drive 0  --wifi-region off
```

tioneb22, 2022-07-03:
Hey @openoms , I tried running the above from a Debian 11 VM but I get a "403 Forbidden" when trying to access the nginx page. I have a 32GB OS drive on /dev/sda1 and my blockchain drive (/dev/sdb1) is mounted to /mnt/hdd. I run the script and nothing gets populated in the /mnt/hdd folder. Not sure what I'm missing... any help is greatly appreciated.

openoms, 2022-07-03:
connect with ssh to your VM

tioneb22, 2022-07-05:
> connect with ssh to your VM

I was connected via SSH. I couldn't figure out how to get the scripts to generate data inside /mnt/hdd. I had mounted it manually and added an entry inside of /etc/fstab.

openoms, 2022-07-05:
when you boot the OS image it simply wait for a second disk to be attached. Should work on the fly, but try restarting the VM if not picked up automatically.