raspiblitz/raspiblitz #2537: SSL certificate problems.

SomeBTChomer, 2021-09-13:
Hello! I created an IP2TOR bridge for my btcpayserver and tried to set up the free SSL certificate option.
At first, none of the two options was working at all. After cancelling those and trying again, the duckdns option works, or at least it says it does. Checking the subscriptions list, raspiblitz does return an ok https certificate (it complained about an invalid one in the first tries.) (I chose https for an ip2tor bridge in the setup, fwiw).

Now:
I can reach my btcpayserver on the ip provided by the IP2TOR-Bridge.
I can reach my btcpayserver on the duckdns domain, but it's not a https connection, which leads to woocommerce not being able to pair with the server. Prompt is: "Connection timed out after 10000 milliseconds."

I tried rebooting, reinstalling the certificate and updating my tor instance to no avail. Maybe someone more knowledgeable can chime in to help me get this btcserver up and running.

Thanks in advance.
Cheers

openoms, 2021-09-13:
See if related: https://github.com/rootzoll/raspiblitz/issues/2533

IP2Tor is not working for the Woocommerce as it is now since it would require using the default HTTPS port 443 and IP2Tor assigns a custom one instead. 
This can be solve until the Woocommerce plugin allows specifiying a custom port. 
Consider opening an feature request there.

SomeBTChomer, 2021-09-13:
Thanks for your response. Issue #2533 is not related. SD card is fine and my raspiblitz is working as usual.

That's a real letdown if it can't be solved to work with woocommerce, I've built my website already, so switching to another shop provider now is a path I'd rather want to avoid.

Wordpress plugins can be modified afaik, but I can't code, else I'd just try pulling this plugin and modifying it to accept a custom port.

So it's either modify the plugin or get a vps for my btcpayserver.

Either way, thanks for the help.

openoms, 2021-09-13:
#2533 is not related to the SDcard, but an IP2Tor service downtime.

You could use and SSH tunnel to a minimal VPS from your Blitz, but that is a  highly custom solution and requires more technical involvement than a dedicated VPS for BTCPayserver.

SomeBTChomer, 2021-09-13:
Whoops, read that wrong. (I think) the downtime can't be the culprit, since the IP2tor service is working fine.

I'd not be comfortable setting up an ssh tunnel directly to my node, so running the btcpayserver on a different machine seems like the best option for me right now. Thanks!

openoms, 2021-09-13:
Opened an issue to support this use-case since it is a common request: https://github.com/btcpayserver/woocommerce-plugin/issues/66

SomeBTChomer, 2021-09-13:
Great! I'm closing this issue since it can't be resolved atm anyway.

closed 2021-09-13 by SomeBTChomer