raspiblitz/raspiblitz #2539: just moving this off to the side.....

DrBlackross, 2021-09-13:
ok, install packer, vagrant, virtualbox (from the readme in the amd64 directory).

-----------------------------------------------------------------------------------------------------------------------------------------------------------------------
4. virtualbox: `brew cask install virtualbox` (maybe re-run after you changed the requested secruity permission to be successfull)

```
cd alternative.platforms/amd64/packer
packer build raspiblitz.json
vagrant box add --force rotzoll/raspiblitz output/raspiblitz.box
cd ../../..
```

With the base box built, you can start a development environment with `vagrant up` and login with `vagrant ssh`.
-----------------------------------------------------------------------------------------------------------------------------------------------------------------------

my output is below

drblackross@TEST:~/raspiblitz-1.7.0/alternative.platforms/amd64/packer$ packer build raspiblitz.json
virtualbox-iso output will be in this color.

1 error(s) occurred:

* The iso_checksum_type must be specified.                                      <----------------- stuck here
drblackross@TEST:~/raspiblitz-1.7.0/alternative.platforms/amd64/packer$ ls
http  raspiblitz.json  scripts

-----------------------------------------------------------------------------------------------------------------------------------------------------------------------

drblackross@TEST:~/raspiblitz-1.7.0/alternative.platforms$ nano display.alternatives.sh 
drblackross@TEST:~/raspiblitz-1.7.0/alternative.platforms$ chmod 755 ./display.alternatives.sh 
drblackross@TEST:~/raspiblitz-1.7.0/alternative.platforms$ ./display.alternatives.sh 
Detect Base Image ...
OK running ubuntu
-----------------------------------------------------------------------------------------------------------------------------------------------------------------------

(SOB JUST now see it lol)
drblackross@TEST:~/raspiblitz-1.7.0$ ./build_sdcard.sh 

*****************************************
* RASPIBLITZ SD CARD IMAGE SETUP v1.7   *
*****************************************
For details on optional parameters - see build script source code:
1) will use NO-INTERACTION --> 'false'
2) will use FATPACK --> 'false'
3) will use GITHUB-USERNAME --> 'rootzoll'
4) will use GITHUB-BRANCH --> 'dev'
5) will use DISPLAY-CLASS --> 'hdmi'
6) will use TWEAK-BOOTDRIVE --> 'true'
7) will use WIFI --> 'US'
X) will use CPU-ARCHITECTURE --> 'x86_64'
X) will use OPERATINGSYSTEM ---> 'ubuntu'
Do you agree with all parameters above? (yes/no) yes
Building RaspiBlitz ...

*** INSTALL TOR BY DEFAULT ***

DrBlackross, 2021-09-13:
ok, it crashed (LOL)

lnd-linux-amd64-v0.12.1-beta.tar.gz                   100%[======================================================================================================================>]  17.87M  7.09MB/s    in 2.5s    

2021-09-13 18:35:42 (7.09 MB/s) - ‘lnd-linux-amd64-v0.12.1-beta.tar.gz’ saved [18736224/18736224]

- download done
- checksum test
Valid SHA256 checksum(s) should be: 4a6b258e580cc43c0ad853adf9e2432aa125c2f5ac3ad7b7947d06c273f7d1a1
9f06276f311155d5436bbe964bbfbf43c6aff75ea93552320abe22bf6908fbd7
c3a79f743aa93f5c3eddd3e055aad0ef8773a6b1011756ad3c6e120453551bbd
Downloaded binary SHA256 checksum: 4a6b258e580cc43c0ad853adf9e2432aa125c2f5ac3ad7b7947d06c273f7d1a1

****************************************
OK --> VERIFIED LND CHECKSUM IS CORRECT
****************************************

- install LND binary
- OK install of LND done

*** raspiblitz.info ***
cat: /home/admin/raspiblitz.info: No such file or directory

**********************************************
BASIC SD CARD BUILD DONE
**********************************************

Your SD Card Image for RaspiBlitz is ready (might still do display config).
Take the chance & look thru the output above if you can spot any errors or warnings.

IMPORTANT IF WANT TO MAKE A RELEASE IMAGE FROM THIS BUILD:
1. login fresh --> user:admin password:raspiblitz
2. run --> ./XXprepareRelease.sh

*** ADDITIONAL DISPLAY OPTIONS ***
- calling: blitz.display.sh set-display hdmi
# blitz.display.sh set-display hdmi 
# old()
# new(hdmi)
/home/admin/config.scripts/blitz.display.sh: line 550: uninstall_: command not found
# nothing to install - hdmi is the default/clean mode
# /mnt/hdd/raspiblitz.conf does not exists (yet) - change is just part of raspiblitz.info
sed: can't read /home/admin/raspiblitz.info: No such file or directory

DrBlackross, 2021-09-13:
kinda weird its looking for minecraft

Package 'oracle-java12-jdk' is not installed, so not removed
Package 'oracle-java7-installer' is not installed, so not removed
Package 'oracle-java8-installer' is not installed, so not removed
Package 'oracle-java9-installer' is not installed, so not removed
Package 'oracle-java10-installer' is not installed, so not removed
Package 'oracle-java12-installer' is not installed, so not removed
E: Unable to locate package nuscratch
E: Unable to locate package minecraft-pi
Reading package lists... Done
Building dependency tree       
Reading state information... Done

DrBlackross, 2021-09-22:
my outcomes

sudo bash build_sdcard.sh true false rootzoll dev hdmi true false amd64 ubuntu
(and any combinations of the above won't really work that well.... )

i always have to change config.scripts/blitz.litecoin.sh

  # set version (change if update is available)
  litecoinSHA256="ca50936299e2c5a66b954c266dcaaeef9e91b2f5307069b9894048acf3eb5751"
  # download
  binaryName="litecoin-${litecoinVersion}-x86_64-linux-gnu.tar.gz"

itll will mostly work, but i have to have in fstab UUID=(of big HD) mounted in fstab before rebooting and beginning raspiblitz install, btrfs works.... blockchains sync (lnd and ltc), i haven't tried BTC yet, i don't want to wait a week for the blockchain to sync.

soooo far the dietpi hc2 version is running ltc fine (give it a week)

cstenglein, 2023-06-07:
@DrBlackross  is this still an issue with 1.9.0?

Closing this for now, please re-open if it still persists.

closed 2023-06-07 by cstenglein