raspiblitz/raspiblitz #2540: [Update] Bitcoin Core v22.0.0

openoms, 2021-09-14:
Manual update script with PGP verification: https://github.com/openoms/bitcoin-tutorials/tree/master/raspiblitz.updates#v220

WIll record any issues here.

rootzoll, 2021-09-14:
Can you also merge this into the build script .. I will do a new build for the 1.7.1RC2 anyway.

openoms, 2021-09-14:
ok - updated the build script.

the bitcoin.update.sh will need to be refactored due to the divergent verification methods

rootzoll, 2021-09-14:
OK let me know when ready to test

openoms, 2021-09-14:
for the custom upgrades we should probably just not allow downgrades below 22 so thee will be only one way to verify.

Alternatively get the user to paste a custom PGP keyID

rootzoll, 2021-09-14:
Yeah I think thats a fair trade off.

SkabbMask, 2021-09-16:
Also updated manually with @openoms script :D ran smooth, no problems!

openoms, 2021-09-17:
https://github.com/rootzoll/raspiblitz/commit/11a67de7cb2970b09d00700e9f8649d0af40a59f

the Bitcoin update script is also changed to use the new community signing process.
Can paste a custom signing pubkey fingerprint with the fallback to W. J van der Laan's.
No support for downgrading below 22.0.