raspiblitz/raspiblitz #2542: remove litecoin

nyxnor, 2021-09-14:
LOL, it is not being taking care of and it carried on, when it should not be.

rootzoll, 2021-09-14:
I also miss time to test it for new releases ... maybe like with other abandon features/services we should start to remove it from the SSH setup menus with v1.8.0 ... and if than somebody commits again to maintain it we can reconsider before removing it completly.

openoms, 2021-09-14:
ACK

(if we want to be bold and busy can consider adding Liquid instead)

nyxnor, 2021-09-14:
> (if we want to be bold and busy can consider adding Liquid instead)

Pls be a joke

jasenm69, 2021-09-26:
i want to build litecoin nodes for the litecoin fam. these are real ppl i know in the real world, good folks and they love ltc. please do not remove this i plan on adding volume here

taoisimo, 2021-09-26:
I second @jasenm69. Don't give up on Litecoin support. I sense that more than a few folks want to get on board with setting up LN nodes shortly. A lot of chatter trending on Twitter recently.

Will benefit everyone to continue supporting Litecoin. Thanks for considering.

nyxnor, 2021-09-26:
Can any of you commit to maintain the litecoin implementation?
> rootzoll: and if than somebody commits again to maintain it we can reconsider before removing it completly.

taoisimo, 2021-09-26:
Admittedly, no. I cannot. I don't want to burden anyone with upkeep if only a handful of folks are taking advantage. I was about to install this later in the week, but I'm only one person and I have no idea how to perform maintenance.

Is there a way to archive everything in case someone wanted to make that commitment in the future?

jasenm69, 2021-09-26:
Is there a way to grab whats build for litecoin . I'm a noob but ill pull together a team and dedicate to building it out.

Sent from ProtonMail mobile

-------- Original Message --------

jasenm69, 2021-09-26:
I'm in my only problem is I'm not very experienced. I know a solid crew of litecoin OG'S we can pick up the slack.

Sent from ProtonMail mobile

-------- Original Message --------

jasenm69, 2021-09-26:
Ove been chattering it up on Twitter to the litecoin family. However I can get a ltc lightning node built i want to.

Sent from ProtonMail mobile

openoms, 2021-09-26:
In my opinion the most practical would be to simply maintain a fork for litecoin. I have no intention to test it so will likely break with further changes.
Litecoin users can either continue using the last compatible release or improve on it as a separate project.

nyxnor, 2021-09-26:
agree with Openoms.

Also about this
> Is there a way to grab whats build for litecoin . I'm a noob but ill pull together a team and dedicate to building it out.

Yes, that is how git repositories works, they have source control, it will always be "archived".

jasenm69, 2021-09-26:
Good enough, ill have to find so.e more experienced ppl to help me out. Idk how to just fork it.

Sent from ProtonMail mobile

-------- Original Message --------

nyxnor, 2021-09-26:
[image]

Just click on the fork

jasenm69, 2021-09-26:
Great thanks, so what all is involved with maintaining it?

Sent from ProtonMail mobile

-------- Original Message --------

nyxnor, 2021-09-26:
Rephrase that

losh11, 2021-09-27:
Against removal of Litecoin from Raspiblitz. Is that an option here?
I know a lot of people, myself included who use Litecoin for Raspiblitz, so I have no idea why it's being removed.

openoms, 2021-09-27:
Just to be clear it is not being removed from the existing versions, but there are no resources to maintain and test updates.
It will be not offered as an option in the future releases due to a lack of maintainers running litecoin.

jasenm69, 2021-09-27:
I forked this project, if you are a developer or coder anything we can build out the litecoin side all we want. I forked it to preserve it, I do not have the skill to build it out. Spread the word with litecoiners im building out nodes for the ones I know.

Sent from ProtonMail mobile

-------- Original Message --------

rootzoll, 2021-09-30:
Hi. I just removed litecoin already from the v1.7.1 because I am totally lacking time to implement and test that integration in the refactored setup process that we needed to prepare for the WebUI.

I also recommend to fork the project if anybody wants to continue litecoin support. Basically it would still be possible to also have it in RaspiBlitz itself even in the future - but that would need more resources that I personally lack at the moment - especially now with the v1.7.1 getting closer.

closed 2021-10-09 by rootzoll