raspiblitz/raspiblitz #2543: customized installation

nyxnor, 2021-09-14:
installation, not build. Build should be minimal, installation user choice.

Bitcoin Network should be checklist to select one or more options:
- [ ] Bitcoin
- [ ] LND
- [ ] C-Lightning

Chain should be checklist to select one or more options:
- [ ] Main
- [ ] Test
- [ ] Signet

Network should be checklist to select one or more options
- [ ] Public IP
- [ ] tor
- [ ] i2p

nyxnor, 2021-09-14:
i2p PR will conflict with tor PR because of this https://github.com/rootzoll/raspiblitz/blob/dd94f8cc9b6e15fa46aaca7018a21f20f2dd39d2/home.admin/config.scripts/internet.tor.sh#L58

nyxnor, 2021-09-21:
not for this release, 1.8.1 it is.

rootzoll, 2024-08-06:
For RaspiBlitz I think we stay the way it is right now .. there is a NixBlitz project in the works that will further go down the road of System by Confing.

closed 2024-08-06 by rootzoll