raspiblitz/raspiblitz #2544: bitcoind service is NOT RUNNING!

TheTiny, 2021-09-14:
Ich Schwierigkeiten mich mich mit dem Passwort C zu authentifizieren.
Ich sehe in den Logs "Loading Block Index"
Tail auf /mnt/hdd/bitcoin/debug.log zeigt, dass er immer noch am Syncen ist.
Aber gleichzeitig verrät mir df -h, dass der Speicherverbrauch der Blockchain nicht größer wird.
Ich hab so den Verdacht, da stimmt was grundsätzlich nicht.
```
/home/admin/XXdebugLogs.sh | nc termbin.com 9999
nginx: the configuration file /etc/nginx/nginx.conf syntax is ok
nginx: configuration file /etc/nginx/nginx.conf test is successful
https://termbin.com/gcl0o
```

TheTiny, 2021-09-14:
Thank you for the quit reply!
Thats what I had hoped to hear. Buy if everything works as expected, shouldn't the /mnt/hdd slowly but surely fillup with more blockchain to store?

nyxnor, 2021-09-14:
yes, just watch the logs
```
sudo tail -fu /mnt/hdd/bitcoin/debug.log
```

TheTiny, 2021-09-14:
I do! I can tell there is work going on.
I am just irritiated that I dont see changes on the disk space calling "df -h"

TheTiny, 2021-09-15:
Its running normal again and the disk starts to grow.
Case closed...

closed 2021-09-15 by TheTiny