raspiblitz/raspiblitz #2547: Dietpi automated script

DrBlackross, 2021-09-18:
https://github.com/rootzoll/raspiblitz/blob/v1.7/alternative.platforms/dietpi/Odroid_HC1_XU4.md

COOL..... but Automation_Custom_Script.sh

root@DietPi:~# tail -n1000 -f /var/tmp/dietpi/logs/dietpi-automation_custom_script.log
--2021-09-18 05:23:15--  https://raw.githubusercontent.com/rootzoll/raspiblitz/v1.6/build_sdcard.sh
Resolving raw.githubusercontent.com (raw.githubusercontent.com)... 185.199.109.133, 185.199.110.133, 185.199.111.133, ...
Connecting to raw.githubusercontent.com (raw.githubusercontent.com)|185.199.109.133|:443... connected.
HTTP request sent, awaiting response... 200 OK
Length: 33866 (33K) [text/plain]
Saving to: ‘build_sdcard.sh’

     0K .......... .......... .......... ...                  100% 7.01M=0.005s

2021-09-18 05:23:15 (7.01 MB/s) - ‘build_sdcard.sh’ saved [33866/33866]

*****************************************
* RASPIBLITZ SD CARD IMAGE SETUP v1.6   *
*****************************************

*** CHECK INPUT PARAMETERS ***
will use code from branch --> 'master'
will use code from user --> 'rootzoll'
will use root partition --> '/dev/mmcblk0p2'
will activate LCD screen --> 'true'
will disable wifi --> 'false'
will use Wifi country --> 'US'
Do you wish to install Raspiblitz branch master? (yes/no) ^C
root@DietPi:~# tail -n1000 -f /var/tmp/dietpi/logs/dietpi-automation_custom_script.log
--2021-09-18 05:23:15--  https://raw.githubusercontent.com/rootzoll/raspiblitz/v1.6/build_sdcard.sh
Resolving raw.githubusercontent.com (raw.githubusercontent.com)... 185.199.109.133, 185.199.110.133, 185.199.111.133, ...
Connecting to raw.githubusercontent.com (raw.githubusercontent.com)|185.199.109.133|:443... connected.
HTTP request sent, awaiting response... 200 OK
Length: 33866 (33K) [text/plain]
Saving to: ‘build_sdcard.sh’

     0K .......... .......... .......... ...                  100% 7.01M=0.005s

2021-09-18 05:23:15 (7.01 MB/s) - ‘build_sdcard.sh’ saved [33866/33866]

*****************************************
* RASPIBLITZ SD CARD IMAGE SETUP v1.6   *
*****************************************

*** CHECK INPUT PARAMETERS ***
will use code from branch --> 'master'
will use code from user --> 'rootzoll'
will use root partition --> '/dev/mmcblk0p2'
will activate LCD screen --> 'true'
will disable wifi --> 'false'
will use Wifi country --> 'US'
Do you wish to install Raspiblitz branch master? (yes/no) 

how do i pass a yes or no /\

DrBlackross, 2021-09-18:
(i have a ton of weird hardware to test raspiblitz on, lol)

DrBlackross, 2021-09-18:
found this in the downloaded build_sdcard.sh
  noInteraction="false"
and set it to true

DrBlackross, 2021-09-18:
think i got it with the automation script (which is still cool)

wget https://raw.githubusercontent.com/openoms/raspiblitz/DietPi/build_sdcard.sh  && sudo bash build_sdcard.sh true false rootzoll v1.7 headless true false arm dietpi

DrBlackross, 2021-09-18:
did have to run 
pip install grpcio grpcio-tools googleapis-common-protos
to progress past the lnd wallet password screen

BUT COOL it worked!
(so far till i break it lol)

DrBlackross, 2021-09-18:
bootstrapping lintecoin (i know, bad)
BUT, @openoms is there a way to roll this into raspiblitz?
https://github.com/LLeny/lncli-curses
i can't find go in the dietpi setup on my hc1

openoms, 2021-09-18:
would just need to install under the bitcoin user, but don't have high hopes, that software  wasn't updated since 3 years.

DrBlackross, 2021-09-19:
[image]

going to let the litecoin bootstrap work first then run bonus.go.sh

but so far so good....

DrBlackross, 2021-09-19:
all sync'd up.... but now i get 
[image]

which is weird, cause i ping out, it sync's just fine, but gives me a rpc error

DrBlackross, 2021-09-19:
(i do love this comment though)

 Do you want to set the fixed DNS 1.1.1.1 by Cloudflare (they 
claim they provide privacy) for your RaspiBlitz and reboot?

DrBlackross, 2021-09-20:
ok downloading the bootstrap then installing go after, well it kinda broke LOL
but, doing a standard sync is working better

                                                                               ┌──────────────── litecoin / main ──────────────────┐
                                                                               │  Waiting for final Blockchain Sync                │  
                                                                               │  Progress: 96.9740864307082800 %                  │  
                                                                               │  Please wait - this can take some long time.      │  
                                                                               │  Its OK to close terminal and ssh back in later.  │  
                                                                               └───────────────────────────────────────────────────┘  
                                                                                 

(lot more accurate, lol)

openoms, 2022-04-04:
Closing due to inactivity. Dietpi is not tested for a good while now.

closed 2022-04-04 by openoms