raspiblitz/raspiblitz #2548: [Update] Electrs v0.9

openoms, 2021-09-18:
https://github.com/romanz/electrs/blob/v0.9.0-rc1/RELEASE-NOTES.md

v0.9.0-rc1 is out

it comes with a database format change so will reindex for a few hours.
Will need to see how long it takes in the RPi before it is usable again.

openoms, 2021-09-19:
New install of v0.9.0-rc1 on an RPi4 built ok in ~40 minutes.
Now indexing.
Does it need the reboot after installing?

openoms, 2021-09-19:
Got an error:
```
Sep 19 20:05:55 tblitz electrs[21430]: [2021-09-19T19:05:55.276Z DEBUG electrs::p2p] got 2000 new headers                                                                                                         
Sep 19 20:05:55 tblitz electrs[21430]: [2021-09-19T19:05:55.280Z INFO  electrs::index] indexing 2000 blocks: [211201..213200]                                                                                     
Sep 19 20:05:55 tblitz electrs[21430]: [2021-09-19T19:05:55.280Z DEBUG electrs::p2p] loading 10 blocks
Sep 19 20:05:55 tblitz electrs[21430]: [2021-09-19T19:05:55.280Z TRACE electrs::p2p] send: GetData([WitnessBlock(00000000000003d56833b4af2f0e89e05adda37512f6f221d12d1ae5b0c71978), WitnessBlock(00000000000003cd6f9db462781341f221a94cb223926d098959ba2bd5c57605), WitnessBlock(00000000000002453ca221807941e44343fbba729caf8b43f8303f0ad6db6097), WitnessBlock(0000000000000354e8e0f3128f7d1f07ae62b989e1d0600c4742c07cd42ef77b), WitnessBlock(00000000000001eb7dfea1dc3d56fb5adfa74a272bbf0f35eb6ae14de30c8d01), WitnessBlock(00000000000002282b68cc9a12561c80f9a0101334af108994f4cf4b242b7376), WitnessBlock(00000000000001efcefe7deb3c02a32fe5d5428be37a50bd1c8e40b8c7cf8504), WitnessBlock(000000000000011bbdc96f199075097ba212ac81b7a06e19fcd4870762da7a75), WitnessBlock(0000000000000463bfedf09b852474c255b9cc60556316ecdf94d873c3c3e09b), WitnessBlock(000000000000022f5d525efaaa6c37d4be525bfd8cf0fce2c4869e0def0f09de)])
Sep 19 20:05:55 tblitz electrs[21430]: [2021-09-19T19:05:55.332Z INFO  electrs::db] closing DB at /mnt/hdd/app-storage/electrs/db/bitcoin                                                                         
Sep 19 20:05:55 tblitz electrs[21430]: Error: initial sync failed
Sep 19 20:05:55 tblitz electrs[21430]: Caused by:
Sep 19 20:05:55 tblitz electrs[21430]:     0: failed to get block 00000000000003d56833b4af2f0e89e05adda37512f6f221d12d1ae5b0c71978                                                                                
Sep 19 20:05:55 tblitz electrs[21430]:     1: p2p failed to recv
Sep 19 20:05:55 tblitz electrs[21430]:     2: I/O error: unexpected end of file
Sep 19 20:05:55 tblitz systemd[1]: electrs.service: Main process exited, code=exited, status=1/FAILURE
Sep 19 20:05:55 tblitz systemd[1]: electrs.service: Failed with result 'exit-code'.
```

openoms, 2021-09-19:
this is weird, when looking the bitcoinlogs 
`/mnt/hdd/bitcoin/debug.log` was not available despite the `bitcoind.service` had no errors.

`sudo systemctl restart bitcoind`
brought back the `debug.log` and the electrs synced continued.

n.b.: the `tbitcoind` and `sbitcoind` services are running too

openoms, 2021-09-19:
The error above happened again at a different block. Bitcoind is working normally this time and electrs restarts to no avail.

Will sync to the latest scripts from `dev`, restart and see how it goes.

openoms, 2021-09-20:
Likely cause:
`maxuploadtarget=5000` in the `bitcoin.conf`

The solution is to add:
`whitelist=download@127.0.0.1`
to bitcoin.conf when electrs is installed

rootzoll, 2021-09-20:
This is the status auf Electrs after updating RaspiBlitz to Electrs v0.9 ... so it says `initialSynced=1`while `syncProgress=0%` .. so can it make sense to reset the "/mnt/hdd/app-storage/electrs/initial-sync.done"?

[image]

openoms, 2021-09-21:
Now it is synced after ~10 h indexing + compaction, but cycling this:
```
Sep 21 08:09:59 tblitz systemd[1]: Stopped Electrs.
Sep 21 08:09:59 tblitz systemd[1]: Started Electrs.
Sep 21 08:09:59 tblitz electrs[17255]: Config { network: Bitcoin, db_path: "/mnt/hdd/app-storage/electrs/db/bitcoin", daemon_dir: "/home/electrs/.bitcoin", daemon_auth: UserPass("raspibolt", "<sensitive>"), daemon_rpc_addr: V4(127.0.0.1:8332), daemon_p2p_addr: V4(127.0.0.1:8333), electrum_rpc_addr: V4(0.0.0.0:50001), monitoring_addr: V4(127.0.0.1:4224), wait_duration: 20s, index_batch_size: 10, index_lookup_limit: Some(200), auto_reindex: true, ignore_mempool: false, sync_once: false, server_banner: "Welcome to electrs v0.9.0-rc1 - the Electrum Rust Server on your RaspiBlitz", args: [] }
Sep 21 08:09:59 tblitz electrs[17255]: [2021-09-21T07:09:59.991Z INFO  electrs::metrics::metrics_impl] serving Prometheus metrics on 127.0.0.1:4224
Sep 21 08:10:00 tblitz electrs[17255]: [2021-09-21T07:10:00.140Z INFO  electrs::db] "/mnt/hdd/app-storage/electrs/db/bitcoin": 141 SST files, 31.648221717 GB, 3.942450137 Grows
Sep 21 08:10:05 tblitz electrs[17255]: [2021-09-21T07:10:05.388Z INFO  electrs::chain] loading 701527 headers, tip=0000000000000000000a1dbe25fd3236403c74be52c821cfcaf76081f0ef0429
Sep 21 08:10:10 tblitz electrs[17255]: [2021-09-21T07:10:10.281Z INFO  electrs::chain] chain updated: tip=0000000000000000000a1dbe25fd3236403c74be52c821cfcaf76081f0ef0429, height=701527
Sep 21 08:10:41 tblitz electrs[17255]: [2021-09-21T07:10:41.427Z INFO  electrs::server] serving Electrum RPC on 0.0.0.0:50001
Sep 21 08:11:01 tblitz electrs[17255]: [2021-09-21T07:11:01.435Z INFO  electrs::server] stopping Electrum RPC server
Sep 21 08:11:01 tblitz electrs[17255]: [2021-09-21T07:11:01.435Z WARN  electrs::thread] tip_loop thread failed: JSON-RPC error: transport error: Didn't receive response data in time, timed out.
Sep 21 08:11:01 tblitz electrs[17255]: [2021-09-21T07:11:01.435Z INFO  electrs::db] closing DB at /mnt/hdd/app-storage/electrs/db/bitcoin
Sep 21 08:11:01 tblitz systemd[1]: electrs.service: Succeeded.
Sep 21 08:12:01 tblitz systemd[1]: electrs.service: Service RestartSec=1min expired, scheduling restart.
Sep 21 08:12:01 tblitz systemd[1]: electrs.service: Scheduled restart job, restart counter is at 180.
Sep 21 08:12:01 tblitz systemd[1]: Stopped Electrs.
```

openoms, 2021-09-21:
Seems like a communication problem with bitcoind (which is running without errors):
```
WARN  electrs::thread] tip_loop thread failed: JSON-RPC error: transport error: Didn't receive response data in time, timed out.
```
Rebuilding electrs now from the latest master where the rust bitcoin crate is bumped: https://github.com/romanz/electrs/commit/3041e89cd2fb377541b929d852ef6298c2d4e60a

rootzoll, 2021-09-21:
Maybe we wanna stick with the old electrs for the v1.7.1 and push the new one to 1.8.0?

openoms, 2021-09-21:
> Maybe we wanna stick with the old electrs for the v1.7.1 and push the new one to 1.8.0?

yes, might make sense as it also avoids the reindexing after flashing the SDcard. People could update later electively. Downgrading from the rc1 for the release now.

openoms, 2021-09-21:
@rootzoll now the issue with the restarts is solved and Electrs 0.9.0 seems to be significantly more performant compared to the previous database version, also it takes less space now.

There is no question that we should byte the bullet regarding the reindexing the question is just when?

I would like to include electrs v0.9.0 in the blitz v1.7.1 if the stable release is out by then.

Kixunil, 2021-09-21:
FTR @romanz [wants to realease next RC in a ~week](https://github.com/romanz/electrs/issues/490#issuecomment-922505724)

Given that the db will have to be reindexed eventually anyway, doing it sooner will cause the benefits to be experienced for longer time, so doing it ASAP looks the most rational. Perhaps notify the users in some way to only upgrade when they don't need to use electrs for a while.

openoms, 2021-09-21:
ok, PR waiting for the stable release with the status fixed

rootzoll, 2021-09-21:
Merged it in best hopes that a stable realease will come soon ... if not available by end of next week we need to reverse back.

romanz, 2021-09-23:
[`v0.9.0-rc2`](https://github.com/romanz/electrs/releases/tag/v0.9.0-rc2) is released - please let me know if there is any open issue :)

openoms, 2021-09-24:
Had a report that `v0.9.0-rc2` had no restarts so updating for the next raspiblitz RC

openoms, 2021-09-25:
Got this error on TG:
[image]

Changing the config solved it, will need to adapt the default:
`sudo nano /home/electrs/.electrs/config.toml`

add:
```
wait_duration_secs = 10
```

Kixunil, 2021-09-30:
Head up, you may want to implement the logic of deleting old database described in https://github.com/romanz/electrs/blob/master/doc/usage.md#important-changes-from-versions-older-than-090

rootzoll, 2021-09-30:
The final 0.9.0 has been released: https://twitter.com/roman_zeyde/status/1443484929480921090?s=20

rootzoll, 2021-10-04:
Deleting of old database to be done in next release. We just recommend/allow electrs for 1TB drives for some time now so it should be fine to have the old db as a fall back in case of a downgrade for now.

rootzoll, 2021-10-09:
@openoms the install of electrs failed on my latest test with RC3:

```
HEAD is now at 557911e Release 0.9.0
    Updating crates.io index
./config.scripts/bonus.electrs.sh: line 462: 29769 Segmentation fault      sudo -u electrs cargo build --locked --release
```

romanz, 2021-10-09:
Could you please share the build log?
Also, how much RAM is available during the build?

Kixunil, 2021-10-09:
Just stdout. Sefault during *build* is super-strange.

romanz, 2021-10-09:
Could you please retry the build with higher verbosity:
```
cargo build --locked --release -v
```

romanz, 2021-10-09:
Maybe single-threaded building can also help (using `--jobs=1` cargo flag)?

openoms, 2021-10-09:
> @openoms the install of electrs failed on my latest test with RC3:
> 
> ```
> HEAD is now at 557911e Release 0.9.0
>     Updating crates.io index
> ./config.scripts/bonus.electrs.sh: line 462: 29769 Segmentation fault      sudo -u electrs cargo build --locked --release
> ```

Couldn't reproduce this so far. Could it be possible that it was a one off memory error causing this? Did you try again on the same board or on an other one? (would consider labeling that RPi for testing only if this would be the case)

openoms, 2021-10-09:
Reproduced on a complete new install, looking into it:
```
Cloning into 'electrs'...
...
HEAD is now at 557911e Release 0.9.0
    Updating crates.io index
/home/admin/config.scripts/bonus.electrs.sh: line 462: 12171 Segmentation fault      sudo -u electrs cargo build --locked --release
```

Kixunil, 2021-10-09:
Could you run `sudo -u electrs gdb --args cargo build --locked --release` and then type `bt` after it segfaults? (You may need to `apt install gdb`.) This looks like a bug in `cargo` (may or may not be introduced by packaging), not `electrs` but I am capable and willing to help with diagnosing.

Am I correct assuming it's Armbian based on Debian 10?

romanz, 2021-10-10:
https://bugs.debian.org/cgi-bin/bugreport.cgi?bug=918854 may be related.
Could you please run:
```
$ RUST_LOG=debug cargo build --locked --release
```

romanz, 2021-10-10:
We can also check only the dependencies' download (without actually building) by running:
```
$ cargo fetch
```

romanz, 2021-10-10:
If the segfault is HTTP-related, we can increase the relevant log verbosity here:
https://doc.rust-lang.org/cargo/reference/config.html#http

openoms, 2021-10-10:
It is likely a Rust installation issue with `apt`. Affecting an other Rust project as well the same way with segfault during `Updating crates.io index` : https://github.com/rootzoll/raspiblitz/issues/2579

romanz, 2021-10-10:
Sounds good - the only question I have is whether we should pin a specific Rust version (using `--default-toolchain` rustup flag)?
For example, electrs `master` is using [1.41.1](https://github.com/romanz/electrs/blob/master/rust-toolchain), but probably can compile with other Rust versions as well.

openoms, 2021-10-10:
The script installs `rustc 1.55.0`
```
info: default toolchain set to 'stable-aarch64-unknown-linux-gnu'
  stable-aarch64-unknown-linux-gnu unchanged - rustc 1.55.0 (c8dfcfe04 2021-09-06)
Rust is installed now. Great!
```
but electrs does download the specified version anyway: 
```
admin@171rc3:/home/electrs/electrs $  sudo -u electrs /home/electrs/.cargo/bin/cargo build --locked --release || exit 1
info: syncing channel updates for '1.41.1-aarch64-unknown-linux-gnu'
info: latest update on 2020-02-27, rust version 1.41.1 (f3e1a954d 2020-02-24)
info: downloading component 'cargo'
```
Would pinning Rust affect future updates? We can do it anyway as it is only pinned for the `electrs` user

openoms, 2021-10-10:
ok, updated and tested ok (will run again on a clean install)

for anyone on raspiblitz v17.1RC3 to test the latest state run in the CLI:
`patch`
before installing `electrs` (or connecting Fully Noded to CL)

Kixunil, 2021-10-10:
Sorry, forgot that you must run `run` in `gdb`, then it should segfault and then you should use `bt`

Would recommend considering signature verification of Rust if you need to fall back.

According to the bug report @romanz linked it could have something to do with proxy. Do you happen to set Tor proxy as system-wide configuration? Perhaps try to turn it off to see if it helps?

rootzoll, 2021-10-10:
No tested again on the same machine wit the fix of @openoms .. and it installed fine :)
Removing the bug confirmed lable. Will test again on another machine for final testing.
Thanks everybody for the quick response.

closed 2021-10-11 by rootzoll