raspiblitz/raspiblitz #2550: Issues with certificates IP2Tor, BTCPay

Xaspr, 2021-09-18:
Hi,

I struggle with the certificates with an IP2Tor bridge. In Firefox and other browsers I get the message that the certificate is an untrusted self-signed certificate (lnd autogenerated cert by localhost). This is also the case for the local network logins of BTCPay, RTL and Lightning Terminal. 

Both the subscriptions for letsencrypt and the IP2Tor Bridge are active, and I've tried cancelling and subscribing again. If I go to BTCPay Server Info in the Raspiblitz menu I see a warning that there's no HTTPS response. Accessing the hidden service directly via Tor is possible (without warnings).  

Any ideas on how to fix this, also for the logins via the local network?

rootzoll, 2021-09-22:
Just checked for the upcoming v1.7.1 release the IP2Tor & LetsEncrypt features. I had to remove the DYNU domain service for the LetsEncrypt Certificate for now - so if you use that please cancel and try the DUCKDNS one.

Xaspr, 2021-09-23:
Thank you! I'm away for a few weeks now. When I get back I'll check out if 1.7.1 fixes it.

openoms, 2022-07-04:
closing for inactivirty

closed 2022-07-04 by openoms