raspiblitz/raspiblitz #2551: Can't update BTCPay

Xaspr, 2021-09-18:
Hi, 

When I try to update BTCPay in the GUI at 'Server Settings'/'Maintenance' I get the message: "Maintenance feature requires access to SSH properly configured in BTCPayServer configuration." The current BTCPay version is v1.0.7.2. 

I've reinstalled raspiblitz from a fresh SD (v.1.7.0) and on a formatted hard drive, but the problem is there straight away. Is there a guide on how to fix the SSH access, or how to update BTCPay in the raspiblitz implementation manually via command line? 

Many thanks in advance.

openoms, 2021-09-19:
Updates are managed by the raspiblitz scripts not by BTCPay in our setup as we don't use Docker .
The next RasiBlitz release will have the updated version of BTCPay.

openoms, 2021-09-19:
In the dev branch already https://github.com/rootzoll/raspiblitz/commit/9500480dbf8143123d17944ed663a245732114fd

rootzoll, 2021-09-21:
OK .. install looking good. Needs final testing on a node with lightning channels & inbound.

openoms, 2021-09-21:
Updated and tipbox working ok (needs the Tor Browser): 
`http://7tpv3ynajkv6cdocmzitcd4z3xrstp3ic6xtv5om3dc2ned3fffll5qd.onion/i/D1s4iFJmyF319KFU5qaS7B`

As after every update ran into white screens when trying to generate invoices so needed delete the previous ones. Here is how: https://github.com/rootzoll/raspiblitz/issues/1725#issuecomment-726272559

This wasn't reported by anyone else and not sure about the cause.

At some point we should start using postgres instead as in the issue linked.

Xaspr, 2021-09-23:
> Updates are managed by the raspiblitz scripts not by BTCPay in our setup as we don't use Docker .
> 
> The next RasiBlitz release will have the updated version of BTCPay.

I didn't realise the update button was just for Docker implementation. I'll just patch the new release when it's out. Thanks!