raspiblitz/raspiblitz #2553: Re-Check COPY over LAN

rootzoll, 2021-09-19:
Got stuck at ths point on one test run ... after source had completly transferrred the chain and the target was waiting more than 6 hours to continue-

[image]
.

openoms, 2021-09-19:
Tested this successfully by copying from an RPi4 to Ubuntu VM. Also restarting the copy worked after a connection break.

rootzoll, 2021-09-19:
Found the problem in a change I did to the boostrap service.

rootzoll, 2021-09-19:
Process needs to be retested with latest dev branch.

openoms, 2021-09-19:
I wonder if the same issue caused the syncing problem with electrs: https://github.com/rootzoll/raspiblitz/issues/2548
continuing the tests.

rootzoll, 2021-10-09:
re-test worked. closing issue for release.

closed 2021-10-09 by rootzoll