raspiblitz/raspiblitz #2554: No progress on initial blockchain download for Litecoin

getziggy, 2021-09-19:
Hello,
I am running into an issue where at the initial blockchain download for litecoin I am stuck at 0% indefinitely. I reset everything and started again but the same issue persists. Attaching logs: http://termbin.com/72e2
None of the closed issues seems to help. Thanks in advance!

openoms, 2021-09-19:
Just like with bitcoin you should try adding peers manually: https://developer.bitcoin.org/reference/rpc/addnode.html

(if there is any and I have no idea where find them for litecoin)

DrBlackross, 2021-09-23:
i found so far if you shut tor off it will sync, or you can copy the blockchain's blocks and chainstate folders over from another node (or previous node). the other place on here someone posted https://openchains.info/coin/litecoin/peers as a place to get node's to addnode to your conf, i don't see it sync with tor on though so far after adding those nodes in my litecoin.conf.

DrBlackross, 2021-09-23:
>ltc node pi3b+

new LTC node, doesn't sync with tor on.
(thought it was nodes to sync from,  open conf and addnodes)
https://openchains.info/coin/litecoin/peers <--- url for ip's to add to litecoin.conf

added nodes to llitecoin.conf 

addnode=5.39.79.51:9333
addnode=95.216.159.206:9333
addnode=47.112.105.55:9333
addnode=45.77.231.52:9333
addnode=94.130.69.46:9333
addnode=88.99.151.67:9333

restarted litecoind still not sync'ing
i then disabled tor in the services menu (because the node will reboot), and let litecoind sync the blockchain and to snag a list of peers (without tor).... once that was done (full copy of blockchain downloaded and 10+peers) then type menu and enable tor in services, itll reboot again, but this time it will have a full blockchain and a peers list to go by.

it will sort it out.... lol

in short initial sync with tor off for peers list and blockchain, then enable tor, reboot, catch up on the blocks... then create new lnd wallet.

DrBlackross, 2021-09-24:
yea for me its TOR
when i disable TOR in the 'menu'

*** Init LND Service & Start ***
stopping lnd ..
TOR was not selected
Starting LND Service ...
Created symlink /etc/systemd/system/multi-user.target.wants/lnd.service → /etc/systemd/system/lnd.service.

then setup and sync works (semi) fine.... going to leave it sit again for a day without TOR running to sync, and i'll check on it sunday.

getziggy, 2021-10-06:
Solved