raspiblitz/raspiblitz #2555: dialog over whiptail - choose one TUI type

nyxnor, 2021-09-19:
Change all whiptail for dialog.

The code is currently using dialog and whiptail and even if they are similar, dialog is much more customizable and easier to deal with.
If I use JoininBox .dialogrc, it works for dialog yeah, but a huge discrepancy in colors when switching to whiptail for basic info messages.
[image]

rootzoll, 2021-09-19:
At some points whiptail was choosen because (at least on MacOS) it allows copy+paste from the terminal for addresses, etc. So if we go for one I might go for whiptail BUT for the python scripts I just found a dialog library - not a whiptail library. So I think changing to just one would be a nice to have to be "more clean" but for practical reasons we just keep using both.

nyxnor, 2021-09-19:
Ok, did not know dialog was not possible to copy, so another alternative for things like this, instead of using whiptail is printing directly to the terminal, example is when showing the addresses for the services, certificate fingerprint.

Quick search and did not found how to copy from dialogbox :(

rootzoll, 2021-09-19:
Is there any pressing reason not to use both other than "having more consistent code"?

nyxnor, 2021-09-19:
Not much (appearance, more options) but consistent code is kinda relevant to be easier to maintain.

openoms, 2021-09-19:
Printing in terminal is an alternative to using whiptail to copy paste, but I don't see the need to refactor these.

The different colours of JoininBox is intentional to differentiate between the projects so it is obvious which menu is on.

openoms, 2021-09-19:
Also the TUI refers to the Touch User Interface, is that a typo here?

nyxnor, 2021-09-19:
> Printing in terminal is an alternative to using whiptail to copy paste, but I don't see the need to refactor these.

The other problem is that SHA1 for the certificate is not as secure as SHA256 of course, and is pretty much not recommended to verify anything, https://github.com/rootzoll/raspiblitz/issues/2378, but it does not fit the whiptail 80 width.

> The different colours of JoininBox is intentional to differentiate between the projects so it is obvious which menu is on.

I get this point, I was thinking about it.

> Also the TUI refers to the Touch User Interface, is that a typo here?
Based on this https://askubuntu.com/questions/867416/are-there-differences-between-cli-and-tui, I am calling TUI the Terminal User Interface and CLI the Command Line Interface.
TUI being whiptail or dialog
CLI being bitcoin-cli or lncli

Not urgent right now, but consistency is important.

openoms, 2021-09-19:
ok, it is a good point about SHA1. Now it is very inconvenient to verify even with the shorter insecure fingerprints.

I do prefer to show the longer texts in the terminal with an accompanying QRcode.

Same for the SSL fingerprints and Hidden Services.

Aesthetics is not really something we can optimize in the terminal, but can communicate more data better.

closed 2024-08-06 by rootzoll