raspiblitz/raspiblitz #2556: [CLN] LNBits backed by C-lightning

openoms, 2021-09-20:
The configuration changes are simple, but needs testing: https://github.com/lnbits/lnbits/blob/master/docs/guide/wallets.md#clightning

rootzoll, 2021-09-20:
Will be do this like with RTL ... having a LNbits & a cLNbits or will this be a selector within the LNbits options/menus somehow?

Also we should use this refactor to change the LNbits service to a `prestart script` for ad-hoc configuration like RTL or BTC-RPC-Explorer.

rootzoll, 2021-12-04:
OK ... first I tought lets make two LNbits installations one for lnd and one for c-lightning - like we did for RTL. But came to the conclusion that it sjust makes sense to have on LNbits installation that takes just one funding source that can be switched. That way even an already running LNbits with accounts is flexible to just change the funding source. Also its easier with all the IP2Tor to just have one instance - thats less confusing.

So in general - apps like RTL that are needed to manage a node makes sense to have an instance each for LND & c-lightning. But apps that work on-top and can work with both makes sense to have just one instance and make it easy to switch what node should run in the background.

rootzoll, 2021-12-07:
OK - ready for testing in branch: https://github.com/rootzoll/raspiblitz/tree/2556cLNbits

To be looked at on a side note:
- LNbits raspiblitz tag is on an stable but quite old April sate - maybe update? Install still works?
- with LND the copied macaroons from `/mnt/hdd/app-data/lnd/data` are used ... maybe let background process also check if those macaroons changed and need updating?

rootzoll, 2021-12-16:
OK merged now into `dev` branch. On test switching from LND to c-lightning as a LNbits backend I got this error again:
 
```
Dec 16 19:06:37 testblitz bonus.lnbits.sh[1441530]: # OK: prestart finished
Dec 16 19:06:37 testblitz systemd[1]: Started lnbits.
Dec 16 19:06:39 testblitz sh[1441545]: Traceback (most recent call last):
Dec 16 19:06:39 testblitz sh[1441545]:   File "/home/lnbits/lnbits/./venv/bin/hypercorn", line 8, in <module>
Dec 16 19:06:39 testblitz sh[1441545]:     sys.exit(main())
Dec 16 19:06:39 testblitz sh[1441545]:   File "/home/lnbits/lnbits/venv/lib/python3.9/site-packages/hypercorn/__main__.py", line 267, in main
Dec 16 19:06:39 testblitz sh[1441545]:     run(config)
Dec 16 19:06:39 testblitz sh[1441545]:   File "/home/lnbits/lnbits/venv/lib/python3.9/site-packages/hypercorn/run.py", line 34, in run
Dec 16 19:06:39 testblitz sh[1441545]:     worker_func(config)
Dec 16 19:06:39 testblitz sh[1441545]:   File "/home/lnbits/lnbits/venv/lib/python3.9/site-packages/hypercorn/trio/run.py", line 113, in trio_worker
Dec 16 19:06:39 testblitz sh[1441545]:     app = load_application(config.application_path)
Dec 16 19:06:39 testblitz sh[1441545]:   File "/home/lnbits/lnbits/venv/lib/python3.9/site-packages/hypercorn/utils.py", line 110, in load_application
Dec 16 19:06:39 testblitz sh[1441545]:     module = import_module(import_name)
Dec 16 19:06:39 testblitz sh[1441545]:   File "/usr/lib/python3.9/importlib/__init__.py", line 127, in import_module
Dec 16 19:06:39 testblitz sh[1441545]:     return _bootstrap._gcd_import(name[level:], package, level)
Dec 16 19:06:39 testblitz sh[1441545]:   File "<frozen importlib._bootstrap>", line 1030, in _gcd_import
Dec 16 19:06:39 testblitz sh[1441545]:   File "<frozen importlib._bootstrap>", line 1007, in _find_and_load
Dec 16 19:06:39 testblitz sh[1441545]:   File "<frozen importlib._bootstrap>", line 986, in _find_and_load_unlocked
Dec 16 19:06:39 testblitz sh[1441545]:   File "<frozen importlib._bootstrap>", line 680, in _load_unlocked
Dec 16 19:06:39 testblitz sh[1441545]:   File "<frozen importlib._bootstrap_external>", line 790, in exec_module
Dec 16 19:06:39 testblitz sh[1441545]:   File "<frozen importlib._bootstrap>", line 228, in _call_with_frames_removed
Dec 16 19:06:39 testblitz sh[1441545]:   File "/home/lnbits/lnbits/lnbits/app.py", line 11, in <module>
Dec 16 19:06:39 testblitz sh[1441545]:     from .commands import db_migrate, handle_assets
Dec 16 19:06:39 testblitz sh[1441545]:   File "/home/lnbits/lnbits/lnbits/commands.py", line 8, in <module>
Dec 16 19:06:39 testblitz sh[1441545]:     from .db import SQLITE, POSTGRES, COCKROACH
Dec 16 19:06:39 testblitz sh[1441545]:   File "/home/lnbits/lnbits/lnbits/db.py", line 11, in <module>
Dec 16 19:06:39 testblitz sh[1441545]:     from .settings import LNBITS_DATA_FOLDER, LNBITS_DATABASE_URL
Dec 16 19:06:39 testblitz sh[1441545]:   File "/home/lnbits/lnbits/lnbits/settings.py", line 46, in <module>
Dec 16 19:06:39 testblitz sh[1441545]:     WALLET = wallet_class()
Dec 16 19:06:39 testblitz sh[1441545]:   File "/home/lnbits/lnbits/lnbits/wallets/clightning.py", line 48, in __init__
Dec 16 19:06:39 testblitz sh[1441545]:     invoices = self.ln.listinvoices()
Dec 16 19:06:39 testblitz sh[1441545]:   File "/home/lnbits/lnbits/venv/lib/python3.9/site-packages/lightning/lightning.py", line 630, in listinvoices
Dec 16 19:06:39 testblitz sh[1441545]:     return self.call("listinvoices", payload)
Dec 16 19:06:39 testblitz sh[1441545]:   File "/home/lnbits/lnbits/venv/lib/python3.9/site-packages/lightning/lightning.py", line 217, in call
Dec 16 19:06:39 testblitz sh[1441545]:     sock.connect(self.socket_path)
Dec 16 19:06:39 testblitz sh[1441545]: PermissionError: [Errno 13] Permission denied
Dec 16 19:06:39 testblitz systemd[1]: lnbits.service: Main process exited, code=exited, status=1/FAILURE
```

With `CLIGHTNING_RPC=/home/bitcoin/.lightning/bitcoin/lightning-rpc` in LNbits config while
```
admin@192.168.178.50:~ ₿ sudo ls -la /home/bitcoin/.lightning/bitcoin/lightning-rpc
srw-rw---- 1 bitcoin bitcoin 0 Dec 16 10:08 /home/bitcoin/.lightning/bitcoin/lightning-rpc

admin@192.168.178.50:~ ₿ sudo ls -la /home/bitcoin/.lightning/
total 20
drwxr-xr-x  3 bitcoin bitcoin 4096 Dec 14 18:56 .
drwxr-xr-x 10 bitcoin bitcoin 4096 Dec 16 18:47 ..
drwxrwx---  2 bitcoin bitcoin 4096 Dec 16 18:58 bitcoin
-rw-r--r--  1 bitcoin bitcoin  267 Dec 14 18:53 config
-rw-r-----  1 bitcoin bitcoin    7 Dec 16 10:07 lightningd-bitcoin.pid
```

rootzoll, 2021-12-21:
lnbits switched to fastAPI branch - needs update on commit and restest

rootzoll, 2021-12-23:
Opened `2556lnbits` branch

rootzoll, 2021-12-23:
OK change to fastAPI seems to work - merging branch into dev for final testing.

openoms, 2022-02-05:
if during the provision the c-lightning was not run yet the lightning-rpc socket does not exist (or might be recreated)
so lnbits will not be able to read and error out with 
```
File "/home/lnbits/lnbits/venv/lib/python3.9/site-packages/lightning/lightning.py", line 630, in listinvoices
return self.call("listinvoices", payload)
File "/home/lnbits/lnbits/venv/lib/python3.9/site-packages/lightning/lightning.py", line 217, in call
PermissionError: [Errno 13] Permission denied
```
Confirmed:
```
$ sudo ls -la /home/bitcoin/.lightning/bitcoin${clrpcsubdir}/lightning-rpc
srw------- 1 bitcoin bitcoin 0 Feb  5 19:40 /home/bitcoin/.lightning/bitcoin/lightning-rpc

$ groups lnbits
lnbits : lnbits
$ groups bitcoin
bitcoin : bitcoin debian-tor

admin@192.168.122.210:~ ₿   

```

Solved with :
```
sudo chmod 660 /home/bitcoin/.lightning/bitcoin${clrpcsubdir}/lightning-rpc
# also (is this running too early after creating the user?):
sudo /usr/sbin/usermod --append --groups bitcoin lnbits
sudo systemctl restart lnbits
```

```

rootzoll, 2022-02-06:
> if during the provision the c-lightning was not run yet the lightning-rpc socket does not exist (or might be recreated)
so lnbits will not be able to read and error out with

So do I understand that right, that the c-lightning needs to run first (at least once) to create the lightning-rpc socket that lnbits needs on installation? So one solution could be to put the needed configuration into the pre-start script of ln bits?

openoms, 2022-02-06:
Yes, but the limitation of the prestart scripts is that  they are running with the unprivileged user (rightly) so cannot modify ownership or user groups.

openoms, 2022-02-06:
Relaxed the permission of the `lightning-rpc` socket by default to group readable. This make sense for other apps too, like C_lightningREST can be run with it's separate user instead of `bitcoin` as well.

Also added some debug checks to catch if lnbits is not added to the bitcoin group: 04a3a5a4

openoms, 2022-02-09:
do weird.
During install:
```
# Running: 'bonus.lnbits.sh on cl'
*** Add the 'lnbits' user ***
Adding user `lnbits' ...
Adding new group `lnbits' (1003) ...
Adding new user `lnbits' (1003) with group `lnbits' ...
Creating home directory `/home/lnbits' ...
Copying files from `/etc/skel' ...
# add the 'lnbits' user to the 'bitcoin' group
# check user
uid=1003(lnbits) gid=1003(lnbits) groups=1003(lnbits),1002(bitcoin)
# get the github code user(lnbits) branch(3ae6ef25a1fce6fc53d444c9352e4fe7972ed9a3)
Cloning into 'lnbits'...
remote: Enumerating objects: 19193, done.
remote: Counting objects: 100% (921/921), done.
```

the checking afterwards:
```
₿ id lnbits
uid=1003(lnbits) gid=1003(lnbits) groups=1003(lnbits)
```
something must remove it from the `bitcoin` group

openoms, 2022-02-09:
now it works!