raspiblitz/raspiblitz #2557: Error found in logs

velgard1, 2021-09-20:
Hi,
I bought my 4GB Blitz in the official shop end of last year and it worked without big issues so far with V1.7 & BTRFS (USB stick was added by me since it's not part of the package).
When I came back from holiday I had an error message on the screen with the suggestion to try the repair options - the first time this happened.
So I tried to delete only the index - which didn't work, it seems somehow files couldn't be deleted - and later on the whole chain - with the same result. I also turned of the Blitz and I tried to flash the SD card again with V1.7 - didn't work either...
Apart from the error message everything seems to be fine (flash, setup/install, menu access).
Is there a way to check & repair the HDD or another possibility to delete files?
I can provide more detailed information, but first I want to get an idea where to look and what to provide exactly - and maybe there is even a simple solution...
Thanks

rootzoll, 2021-09-21:
The BTRFS is still experimental and further development is pushed down the line because other things have more priority - it can take quite long until BTRFS gets further improved. I would recommend in general to move to a default ext4 HDD format (without an raid).

For that wait until final v1.7.1 is released (beginning of october). Make sure to have from your old install SEED WORDS & StaticChannelBackup File ready. Then try to download the migration file (REPAIR > MIGRATION) and check that its the correct checksum/size. Then shutdown and flash the new v1.7.1 to your sd card and boot it up again (usb stick removed), login and DONT RECOVER instead choose other options and use UPLOAD MIGRATION. You will need to sync the Blockchain again but then you should be good.

velgard1, 2021-09-24:
Thanks, this sounds promising, I'll report if it works after the V1.7.1 release.

velgard1, 2021-10-15:
It's working fine, only needed the migration file. So I'll close the issue.

Only the syncing time is a bit confusing (20% in 24h, and lightning has a separate & different value) - during my inital setup it was much quicker in the beginning and very slow later on, so I hope the speed will not decrease so much this time...

rootzoll, 2021-10-16:
> Only the syncing time is a bit confusing (20% in 24h, and lightning has a separate & different value) - during my inital setup it was much quicker in the beginning and very slow later on, so I hope the speed will not decrease so much this time...

This is due to that in the beginning of the Bitcoin blockchain there were much less transactions and data to validate transactions on. So the the first years of blocks goes quite fast and it sync/validation lows down. I dont think we can do much on this for now.

velgard1, 2021-10-18:
I know, but the first 42% took already 4 days - I assume that this is because I'm running it behind TOR.
During the normal setup you get asked how you want to download the chain and it's said that it's safe & faster to download the chain without TOR and activate it only later on - maybe with the migration file it's simply downloading the chain behind TOR (at theast I was not asked how to download the cain this time).
Another reason could be that the reason is that something with lightning is going on too which is initially not the case...
If TOR could be a reason I would open a new issue to always integrate a dialogue before downloading the chain.

velgard1, 2021-11-04:
It seems to have synced finally now - at around 50% I had weird errors and flashed the SD again, afterwards it seemed to be faster for a while and turned slow again at aroud 80%.
Is such a long time normal? If not is there already a solution or should a new issue be created?
Sadly I face the reboot issue now, I'll comment on this within the right topic...

rootzoll, 2021-11-11:
This sounds not normal .. I will keep this issue open. See if similar reports come in.

velgard1, 2021-11-12:
Sidebar: I was able to solve the loop issue via flashing the SD again - had at least two issues where flashing the actual image again helped to solve the issue. Checked the SD card, it's working fine, only wonder how to check the SSD...
Since especially the last about 10% took so long I wondered if simply the last year that passed since my initial setup could be the issue since it's maybe more compley to validate toe to full mempool etc...

closed 2023-06-07 by cstenglein