raspiblitz/raspiblitz #2560: [Discussion] separate Tor instance for C-lightning and LND with tor.streamisolation

openoms, 2021-09-21:
in v1.7.0 LND had a dedicated Tor instance to solve this issue: https://github.com/rootzoll/raspiblitz/issues/1604#issuecomment-923240959
for which we found that reloading, instead of restarting the tor instance solves the problem too so removed the dedicated `tor@lnd` instance in this PR: https://github.com/rootzoll/raspiblitz/pull/2536 for the sake simplification and to help #2419.

If there are further stability issues it can be reintroduced together with one for C-lightning.

Nb: LND has a `tor.streamisolation` option
https://github.com/lightningnetwork/lnd/blob/9264185f5b3e446b101a09eb371634ac5ba0daf4/sample-lnd.conf#L847
```

; Enable Tor stream isolation by randomizing user credentials for each
; connection. With this mode active, each connection will use a new circuit.
; This means that multiple applications (other than lnd) using Tor won't be mixed
; in with lnd's traffic.
;
; This option may not be used while direct connections are enabled, since direct
; connections compromise source IP privacy by default.
; tor.streamisolation=true
```

have seen some reports that disabling streamisolation helped some stability problems for some users on non-raspiblitz or custom nodes, so just keeping an eye on it.
From v1.7.1 `tor.streamisolation=true` is configured in the lnd.conf and reintroduced on every lnd restart by `lnd.check.sh` so to switch it off for tests would need to comment it out there.

If we would decide to not use `tor.streamisolation` in LND a dedicated Tor instance should be reintroduced to avoid using the same circuit as all other services over Tor. This is also applicable to C-lightning now. It has less significance if there are no Tor services shard publicly (e.g the Lightning is the only public Tor service).

As for Bitcoin Core (the other public Tor service running on the node - see `bitcoin-cli getnetworkinfo`) it uses Tor streamisolation
by default since v0.11.0: https://bitcoin.org/en/release/v0.11.0#privacy-stream-isolation-for-tor

nyxnor, 2021-09-21:
Stream isolation in the torrc
* [tails guide](https://tails.boum.org/contribute/design/stream_isolation/) and [tails torrc](https://gitlab.tails.boum.org/tails/tails/-/blob/master/config/chroot_local-includes/etc/tor/torrc)

* [torbox torrc](https://github.com/radio24/TorBox/blob/d77fe2b2c91d0bc1024bfc19d44d06852738fde1/etc/tor/torrc#L17) - example

* [tor manual](https://2019.www.torproject.org/docs/tor-manual.html.en) - stream isolation

* *SocksPort [address:]port|unix:path|auto [flags] [isolation flags]*

  * Open this port to listen for connections from SOCKS-speaking applications. Set this to 0 if you don’t want to allow application connections via SOCKS. Set it to "auto" to have Tor pick a port for you. This directive can be specified multiple times to bind to multiple addresses/ports. If a unix domain socket is used, you may quote the path using standard C escape sequences. (Default: 9050)

  * NOTE: Although this option allows you to specify an IP address other than localhost, you should do so only with extreme caution. The SOCKS protocol is unencrypted and (as we use it) unauthenticated, so exposing it in this way could leak your information to anybody watching your network, and allow anybody to use your computer as an open proxy.

  * If multiple entries of this option are present in your configuration file, Tor will perform stream isolation between listeners by default. The isolation flags arguments give Tor rules for which streams received on this SocksPort are allowed to share circuits with one another. Recognized isolation flags are:

* *IsolateClientAddr*

  * Don’t share circuits with streams from a different client address. (On by default and strongly recommended when supported; you can disable it with NoIsolateClientAddr. Unsupported and force-disabled when using Unix domain sockets.)
    
* *IsolateSOCKSAuth*

  * Don’t share circuits with streams for which different SOCKS authentication was provided. (For HTTPTunnelPort connections, this option looks at the Proxy-Authorization and X-Tor-Stream-Isolation headers. On by default; you can disable it with NoIsolateSOCKSAuth.)

* *IsolateClientProtocol*

  * Don’t share circuits with streams using a different protocol. (SOCKS 4, SOCKS 5, TransPort connections, NATDPort connections, and DNSPort requests are all considered to be different protocols.)

* *IsolateDestPort*

  * Don’t share circuits with streams targeting a different destination port.

* *IsolateDestAddr*

  * Don’t share circuits with streams targeting a different destination address.

* *KeepAliveIsolateSOCKSAuth*

  * If IsolateSOCKSAuth is enabled, keep alive circuits while they have at least one stream with SOCKS authentication active. After such a circuit is idle for more than MaxCircuitDirtiness seconds, it can be closed.

nyxnor, 2021-09-21:
> If we would decide to not use tor.streamisolation in LND a dedicated Tor instance should be reintroduced to avoid using the same circuit as all other services over Tor. This is also applicable to [...]

Made the commend above of stream isolation on torrc referring to having no need of introcucing a new instance, also possible to configure more socks port in the same torrc. In my point of view, a new instance is only necessary if lnd port closes constantly.

xanoni, 2021-09-21:
> Stream isolation in the torrc

I'm using these flags .. I'm curious if the `lnd` `tor.streamisolation` flag does anything extra if the flags are already set. It shouldn't, as far as I understand.

> Made the commend above of stream isolation on torrc referring to having no need of introcucing a new instance

Remember when I asked a while ago why RaspiBlitz was using a second Tor instance? My thinking was exactly yours. However, then when I tried it myself, that correlated with my channel stability issues disappearing. Thus, I kept it, even though I don't understand why it helps.

xanoni, 2021-09-24:
FYI — https://github.com/lightningnetwork/lnd/discussions/5783

xanoni, 2021-09-24:
@guggero responded, seems we can turn `tor.streamisolation` off even when it's a shared Tor instance.

nyxnor, 2021-12-01:
including stream isolation on
- #2711 https://github.com/nyxnor/raspiblitz/blob/4ca541e498dfce2f24b4875b5e1302a870e0ff1f/home.admin/config.scripts/tor.install.sh#L70

openoms, 2021-12-14:
> including stream isolation on
> 
>     * [[WIP] 2669tor - improve tor configuration #2711](https://github.com/rootzoll/raspiblitz/pull/2711) https://github.com/nyxnor/raspiblitz/blob/4ca541e498dfce2f24b4875b5e1302a870e0ff1f/home.admin/config.scripts/tor.install.sh#L70

@nyxnor does this mean that LND would use a separate Tor circuit in itself, but not one for each of it's connections (unless the `tor.streamisolation=true` is set in the LND.conf)?

The LND `tor.streamisolation=true` is  currently forced in the lnd.check.sh, need to remove it at least for the hybrid mode: https://github.com/rootzoll/raspiblitz/issues/2787#issuecomment-991245694

I propose we should still aim for maximum privacy in case of a Tor-only node, but will configure the LND `tor.streamisolation=false` in hybrid mode.

nyxnor, 2021-12-14:
> - IsolateDestAddr
Don’t share circuits with streams targeting a different destination address.

From this I understand that every connection to a different node will isolate that circuit, meaning different addresses being connected on lighting and btc proxy will use an isolated circuit for each of them.

I think it is possible to see which circuit you are using per connection, but for now I just now how to list the circuits, but not the final destination:
https://stem.torproject.org/tutorials/down_the_rabbit_hole.html
For graphical view with gtk, use the package `onioncircuits`.
For terminal view use `python3-stem`.
```sh
sudo apt install -y python3-stem
## match -i PORT with the ControlPort in the torrc
sudo -u debian-tor tor-prompt -i 9051
```
```py3
for circ in controller.get_circuits():
  print "Cicruit %s, status: %s" % (circ.id, circ.status)
  for i, (fingerprint, nickname) in enumerate(circ.path):
    print " %i. %s (%s)" % (i + 1, fingerprint, nickname)
  print

```

openoms, 2021-12-14:
Colud be interesting to test, but I get:
```
admin@raspberrypi:~ $ sudo -u debian-tor tor-prompt -i 9051
Welcome to Stem's interpreter prompt. This provides you with direct access to
Tor's control interface.

This acts like a standard python interpreter with a Tor connection available
via your 'controller' variable...

  >>> controller.get_info('version')
  '0.2.5.1-alpha-dev (git-245ecfff36c0cecc)'

You can also issue requests directly to Tor...

  >>> GETINFO version
  250-version=0.2.5.1-alpha-dev (git-245ecfff36c0cecc)
  250 OK

For more information run '/help'.

>>> for circ in controller.get_circuits():
...   print "Cicruit %s, status: %s" % (circ.id, circ.status)
File "<console>", line 2
    print "Cicruit %s, status: %s" % (circ.id, circ.status)
                                 ^
SyntaxError: invalid syntax
```

nyxnor, 2021-12-14:
I just copied and pasted from here https://github.com/rootzoll/raspiblitz/issues/2560#issuecomment-993499455
[image]
it show the 3 relays of each cirtuit, fingerprint and nickname, ommitted.

nyxnor, 2021-12-14:
I mailed the creator of stem to see if there is a way to show the circuit final destination, so it would be similar to torbrowser circuit button, but on the terminal. lets see.

nyxnor, 2022-01-04:
The control spec is essential https://gitweb.torproject.org/torspec.git/tree/control-spec.txt#n1637

Seeing your own circuits without installing tor-prompt:
```sh
#!/usr/bin/env sh

circuit_all="$(printf '%s\n%s\n%s\n' "AUTHENTICATE $(hexdump -e '32/1 "%02x""\n"' /run/tor/control.authcookie)" "GETINFO circuit-status" "QUIT" | nc -U /run/tor/control)"
circuit_all="$(printf '%s\n' "${circuit_all}" | sed "/250 OK/d;/250+circuit-status=/d;/250 closing connection/d")"
printf '%s\n' "${circuit_all}" | while IFS="$(printf '\n')" read -r circuit || [ -n "${circuit}" ]; do
  circuit_id="$(printf '%s\n' "${circuit}" | awk '{print $1}')"
  circuit_status="$(printf '%s\n' "${circuit}" | awk '{print $2}')"
  circuit_path="$(printf '%s\n' "${circuit}" | awk '{print $3}')"
  [ -z "${circuit_status}" ] && break
  r=0
  printf '%s\n' "Circuit ${circuit_id}, status: ${circuit_status}"
  for relay in $(printf '%s\n' "${circuit_path}" | tr "," " "); do
    r=$((r+1))
    [ ${r} -gt 3 ] && break
    relay="$(printf '%s\n' "${relay}" | tr -d "$" | tr "~" " ")"
    relay_fingerprint="$(printf '%s\n' "${relay}" | awk '{print $1}')"
    relay_nickname="$(printf '%s\n' "${relay}" | awk '{print $2}')"
    printf '%s\n' "${r}. ${relay_fingerprint} (${relay_nickname})"
  done
  printf "\n"
done
```