raspiblitz/raspiblitz #2563: Specter config needs fixing

rootzoll, 2021-09-21:
Specter installs & starts but the config seems still broken .. these are the results when trying to login:

[image]

seeing those specter logs ... not sure if related:
```
Sep 21 16:22:47 testbrel python3[16484]: [2021-09-21 16:22:47,540] ERROR in wallet_manager: Prevented Trying to update wallet_Manager with broken <BitcoinRpc http://localhost:8332>
Sep 21 16:22:47 testbrel python3[16484]: [2021-09-21 16:22:47,545] INFO in wallet_manager: Specter seems to be disconnected from Bitcoin Core. Skipping wallets update.
Sep 21 16:23:21 testbrel python3[16484]: [2021-09-21 16:23:21,727] INFO in checker: Checker health Checking every 600 seconds now
```

@openoms any idea?

openoms, 2021-09-21:
Ok needed some changes, but now after installation can be configured for the parallel networks with:
```
config.scripts/bonus.specter.sh config testnet
config.scripts/bonus.specter.sh config signet
```
and choose between the nodes within Specter.

[image]

rootzoll, 2021-09-21:
Looks good on first test :) login works now

openoms, 2021-09-22:
Small changes to fix installation in the Ubuntu VM: https://github.com/rootzoll/raspiblitz/commit/90421b1c3b33860ec57a765e4718419828aeefcd