raspiblitz/raspiblitz #2565: pi user logged and executing "/bin/bash /home/admin/00infoLCD.sh"

atef-zayati, 2021-09-21:
Hello,
Sorry if I'm too paranoid. Is this output normal (knowing that I don't have a screen)? Thank you.

```
$ w
 xx:xx:xx up x days, xx:xx,  2 users,  load average: 0.76, 0.83, 0.84
USER     TTY      FROM             LOGIN@   IDLE   JCPU   PCPU  WHAT
pi       tty1     -                DD       xdays  3:16   3:16  /bin/bash /home/admin/00infoLCD.sh
admin    pts/0    x.x.x.x          DD       0.00s  0.29s  0.01s w
```

rootzoll, 2021-09-21:
If you dont have a screen and you dont want the pi user in the background you can (since v1.7.0) call on terminal the command`headless` - this should switch off that logged in pi user after a reboot.

cstenglein, 2023-06-07:
@atef-zayati is this still relevant? Closing for now, if it's still relevant, please re-open

closed 2023-06-07 by cstenglein