raspiblitz/raspiblitz #2576: Fail : BTCPayServer installation is cancelled

nahHoDL, 2021-09-26:
After aborting a BTCPayServer installation mid-way by terminating the terminal window, further attempts to install BTCPayServer fails.

The following message appears
FAIL
BTCPayServer installation was cancelled
Try again from the menue or install from the terminal with:
/home/admin/config.scripts/bonus.btcpayserver.sh on

Re-trying installation from the menu or by terminal is unsuccessful.
The same with installation attempts after reboot.

nyxnor, 2021-09-26:
> Re-trying installation from the menu or by terminal is unsuccessful.

You have to show in the screen way it was unsuccessful

nahHoDL, 2021-09-26:
Thank you for the feedback,

when installing from terminal, this is shown:

# INSTALL BTCPAYSERVER
nginx: the configuration file /etc/nginx/nginx.conf syntax is ok
nginx: configuration file /etc/nginx/nginx.conf test is successful
# Updating the firewall
Skipping adding existing rule
Skipping adding existing rule (v6)
Skipping adding existing rule
Skipping adding existing rule (v6)

# Hidden Service for btcpay
HiddenServiceDir /mnt/hdd/tor/btcpay
HiddenServiceVersion 3
HiddenServicePort 80 127.0.0.1:23002
HiddenServicePort 443 127.0.0.1:23003

Restarting Tor to activate the Hidden Service...

The Tor Hidden Service address for btcpay is:
rg554wccgurhbkzxsxfuwajdxerrvq2gjuexnrkpnm22k54ywxhugytx.onion
use with the port: 80
or the port: 443
# making sure services are not running
perl: warning: Setting locale failed.
perl: warning: Please check that your locale settings:
	LANGUAGE = (unset),
	LC_ALL = (unset),
	LC_CTYPE = "UTF-8",
	LANG = "en_GB.UTF-8"
    are supported and installed on your system.
perl: warning: Falling back to a fallback locale ("en_GB.UTF-8").
adduser: The user `btcpay' already exists.

nyxnor, 2021-09-26:
try setting locale https://askubuntu.com/questions/599808/cannot-set-lc-ctype-to-default-locale-no-such-file-or-directory

openoms, 2021-09-26:
Fixed in `dev` already

for now you an just remove the btcpay user before reinstalling:
`sudo userdel --rf btcpay`

nahHoDL, 2021-09-27:
> Fixed in `dev` already
> 
> for now you an just remove the btcpay user before reinstalling:
> `sudo userdel --rf btcpay`

Many thanks!

sudo userdel --f btcpay

before re-installing worked.

closed 2021-09-27 by nahHoDL