raspiblitz/raspiblitz #2579: [CLN] c-lightning-http-plugin to connect Fully Nodee

openoms, 2021-09-26:
https://github.com/start9labs/c-lightning-http-plugin

Fonta1n3, 2021-09-27:
Thanks for raising this @openoms. Just FYI there are a number of plugins that achieve this (at least there were when I initially researched it), some don't require rust. The above plugin is just what I have used in practice. It would be wonderful if Fully Noded users could access both LND and c-lightning via raspiblitz out of the box :)

openoms, 2021-10-06:
So I gather from https://github.com/rootzoll/raspiblitz/pull/2567#issuecomment-927011217 the QRcode should be (with the CL REST port changed to 9080):

`http://rpcuser:rpcpassword@xxx.onion:9080`

Fonta1n3, 2021-10-11:
> So I gather from [#2567 (comment)](https://github.com/rootzoll/raspiblitz/pull/2567#issuecomment-927011217) the QRcode should be (with the CL REST port changed to 9080):
> 
> `http://rpcuser:rpcpassword@xxx.onion:9080`
> 
> @Fonta1n3 how we set the `rpcuser` here? or does it default to `lightning` ?

I just followed the docs [here](https://docs.rs/crate/c-lightning-http-plugin/0.1.2) for the plugin which makes 8080 the default port. The rpcuser is set as `lightning` by default I beleive.

openoms, 2021-10-12:
Implemented as discussed. Installation and the hidden seervice are functional. Closing for release.