raspiblitz/raspiblitz #2580: CRTL: Error: connect econnrefused 127.0.0.1:6100

ros-mac, 2021-09-27:
This may be reported already and I couldn't see it. CRTL web interface http/https.
I tried sudo ufw allow 6100 and a reload, but didn't work.

Sep 27 11:41:05 raspberrypi node[9487]: --- cl-rest api server is ready and listening on port: 6100 --- warn
Sep 27 11:41:09 raspberrypi node[755]: [2021-09-27T10:41:09.512Z] ERROR: GetInfo => Get Info Error: {"name":"RequestError","message":"Error: connect ECONNREFUSED 127.0.0.1:6100","cause":{"errno":-111,"code":"ECONNREFUSED","syscall":"connect","address":"127.0.0.1","port":6100},"error":{"errno":-111,"code":"ECONNREFUSED","syscall":"connect","address":"127.0.0.1","port":6100},"options":{"url":"https://localhost:6100/v1/getinfo","rejectUnauthorized":false,"json":true,"headers":{},"method":"GET","qs":{},"simple":true,"resolveWithFullResponse":false,"transform2xxOnly":false}}

openoms, 2021-09-27:
check please:
`sudo systemctl status cRTL`
and
`sudo journalctl -fu cRTL`

ros-mac, 2021-09-27:
Thanks! I patched it and rebooted and I had to correct an if statement at line 228 in bonus.rtl.sh
Then I reset the wallet from the menu and restarted the clnrest.service
And it worked!

openoms, 2021-09-27:
Thanks thare are indeed some  extra characters there. Fixing now.

closed 2021-09-27 by openoms