raspiblitz/raspiblitz #2581: tried mucking with 1.7.1rc2EXIT _provision.setup.sh BECAUSE OF ERROR STATE (3)

DrBlackross, 2021-09-27:
# Sourcing /mnt/hdd/raspiblitz.conf 
# RASPIBLITZ CONFIG FILE
raspiBlitzVersion='1.7.1rc2'
lcdrotate=1
lightning=lnd
network=litecoin
chain=main
hostname='MoreBalls'
runBehindTor=on
# Sourcing /var/cache/raspiblitz/temp/raspiblitz.setup 
# RASPIBLITZ SETUP STATE
hostname=MoreBalls
lightning=lnd
network=litecoin
chain=main
setPasswordA=1
setPasswordB=1
setPasswordC=1
passwordA='balls'
passwordB='balls'
passwordC='balls'
# Update /home/admin/raspiblitz.info 
state=provision
message='Starting Provision'
baseimage=raspios_arm64
cpu=aarch64
board=
ramMB=959
ramGB=1
network=litecoin
chain=main
localip='192.168.1.13'
online='1'
dnsworking=1
fsexpanded=1
displayClass=lcd
displayType=
setupStep=0
setupPhase='setup'
fundRecovery=0
hostname=
hddCandidate='sda'
hddGigaBytes=931
hddBlocksBitcoin=0
hddBlocksLitecoin=0
hddGotMigrationData=
SETTING PASSWORD A

OK - password A changed for user pi, root, admin & bitcoin
...
Calling _provision.setup.sh for basic setup tasks ..
EXIT _provision.setup.sh BECAUSE OF ERROR STATE (3)
This can also happen if _provision.setup.sh has syntax errors

openoms, 2021-09-27:
Please give a little more relevant titles to your issues.

Also using basic markdown would improve readability of the logs.

DrBlackross, 2021-09-28:
you know i haven't found a way to turn off the basic markdown thing yet on git hub...  i just copy and paste and it does this formatting

https://guides.github.com/features/mastering-markdown/ <-- doesn't show how to turn it off just shows how to use it (WTH)

anyway, i keep trying to build just a basic clean LTC node, i've done dietpi on hc1 (which worked great for a few days, then locked up, you commented on that thread), RPI 3b+ and 4, also AMD Opteron 6 core with 16gb of ram (couldn't get vagrant (post #1500) to work, but i did get it to install normally with build_sdcard.sh after changing the litecoin url in the blitz.litecoin.sh file with the amd setup) all with totally different drives (480gb, 560gb, and a 1.3tb also all 3 BTRFS together), tried a few different versions of raspiblitz v1.6 up to raspiblitz-v1.7.1rc2-2021-09-23 (most recent i've found)...

it seems (i know this LTC node is "experimental"), but TOR seems to hang the sync for some reason on all different setups, doesn't matter what combo's i try. i disable TOR and litecoind sync's fine, turn on TOR after the blockchain is sync'd and the system locks up (can't ssh in or use a usb keyboard and have to power cycle then login and catch it, to quickly stop the litecoind and lnd services).

on the amd system this worked but like i said i had to change the checksum and the url for litecoind package (not a problem)

sudo bash build_sdcard.sh true false rootzoll dev headless true false amd64 ubuntu <--this works with the AMD system but pulls the wrong litecoind package

X) will use CPU-ARCHITECTURE --> 'x86_64'
X) will use OPERATINGSYSTEM ---> 'ubuntu'

just edit config.scripts/blitz.litecoin.sh after git pull
ca50936299e2c5a66b954c266dcaaeef9e91b2f5307069b9894048acf3eb5751 litecoin-0.18.1-x86_64-linux-gnu.tar.gz

but then still wont sync with TOR on, disable then itll sync, enable TOR system locks... lol

either way the LTC node will fail with TOR on and running, even with bootstrapping the LTC blockchain, i am currently trying 
 bootstrapping with the 9.23.2921 image for kicks on RPI3b+ to see if will work since self sync failed again lol.

currently though the BTC node runs fine no problems and sits in the livingroom humming away, has for about 2 years.

AT this point i am beginning to think its a port conflict with my other node, this all started out as a simple test to slap together a bunch of old hardware that i could rack mount in the basement and walk away from, if it worked, great i'd migrate my other raspiblitz BTC node to other faster hardware with sata and gigabit and walk away ;p (hc1 had 32bit issues lol)

but also my main btc raspiblitz is starting to hit limits, like heat, processing power, ram, etc.... the build_sdcard.sh is great, love you for it... but it took a while for me to figure out how to pass the right arguments after the script to make it pick the right pieces. 

on any setup if btrfs is used it will pick it up, or if a sata drive is mounted as /mnt/hdd in fstab (amd setup and/or with btrfs) it will just roll with it and install raspiblitz (although there was one time i had to edit the fstab after, was weird).

i've done this a total of atleast 25 times, on 7 32gb microsd cards (i ruined a few partition tables (cfdisk wouldn't touch it) and threw them away)

SOOoooo, my only (i guess) question is, would running 2 different raspiblitz systems (one btc and one ltc but both running TOR) 
 would they have a conflict issue, i would think TOR would work around it. 
ltc is 9333 btc is 8333 
and i don't want to pass specific ports to TOR (kinda defeats using it)

BUT, i do love the raspiblitz setup, make life easy, just wish i could make it more widely compatible, and it probably is. (i know this is probably not what raspiblitz was made for, but would be nice to bend it a little, just a little).

openoms, 2021-09-28:
```
put your code between 3 single quotes like this
# there will be no.formatring then, but will appear as a block
```

You can edit your comments.

Running behind Tor means that there will be no conflicts on your local network as all node ia running on separate Tor circuits.

DrBlackross, 2021-09-28:
like now i've left it run again overnight, it was sync'ing... try to log in this afternoon, no route to host. but on the hdmi 'lnd wallet is locked' cursor is flashing bottom left of blue screen (good). but i check the ip traffic on the router for the box nothing after about 20 minutes of logging off the box the night before.

this is when i have to power cycle (which i did) and then ssh in and goto menu and disable TOR, to get control. system reboots and it seems everything works (but doesn't).

i always end up with 

~ sudo tail -n 30 -f /mnt/hdd/litecoin/debug.log

2021-09-28T16:15:11Z init message: Loading block index...
2021-09-28T16:15:11Z Opening LevelDB in /mnt/hdd/litecoin/blocks/index
**2021-09-28T16:15:12Z Fatal LevelDB error: Corruption: error in middle of record: /mnt/hdd/litecoin/blocks/index/000003.log**
2021-09-28T16:15:12Z You can use -debug=leveldb to get more complete diagnostic messages
2021-09-28T16:15:12Z Fatal LevelDB error: Corruption: error in middle of record: /mnt/hdd/litecoin/blocks/index/000003.log
2021-09-28T16:15:12Z : Error opening block database.
Please restart with -reindex or -reindex-chainstate to recover.
2021-09-28T16:15:12Z Aborted block database rebuild. Exiting.
2021-09-28T16:15:12Z Shutdown: In progress...
2021-09-28T16:15:12Z scheduler thread interrupt
2021-09-28T16:15:12Z Shutdown: done

like a handful of blocks in

DrBlackross, 2021-09-28:
then the next steps are to usually repair and wipe the hdd or reset the blockchain without TOR

(lnd is fine btw)

DrBlackross, 2021-09-29:
yea, it kept failing the rpi3b+ (locking up).... 
so just for laughs, slapped the amd A6-7400K with 8gb and a 1tb drive together in the 1ru chassis. installed 18.04 server install (via network install), then ran...

wget --no-cache https://raw.githubusercontent.com/rootzoll/raspiblitz/v1.7/build_sdcard.sh && sudo bash build_sdcard.sh false true rootzoll v1.7 hdmi true true
now.... it worked but when it says reboot and login at the end stopped there and i changed...

nano /home/pi/raspiblitz/home.admin/config.scripts/blitz.litecoin.sh i changed these lines...

  litecoinSHA256="ca50936299e2c5a66b954c266dcaaeef9e91b2f5307069b9894048acf3eb5751"
  
download
  binaryName="litecoin-${litecoinVersion}-x86_64-linux-gnu.tar.gz"

then rebooted, login as admin it setup worked and installed LTC,  BUT there were issues with tor error (111) etc... 
tweaked config files one by one stopping and starting services and rebooting when needed, nothing fixed it. so i put them back they way they were and rebooted and ran 

do-release-upgrade
ubuntu updates the system to 20.04... great. fire up menu and update TOR from the raspiblitz 'menu'
(says it needs the src in the sources.list, they were commented out... un-commented, k done)
try again

BOOM everything works, LMAO
(well so far.... blockchain is at %17.6 and lnd is at %63.5)
now this is just a stock raspiblitz install on originally ubuntu 18.04 server install (x86) that was upgraded to 20.04 after installing raspiblitz... the only thing i needed to do was enable the src sections in sources.list and then use raspiblitz menu to update TOR from source and it compiled and worked

no port forwarded but i did add upnp=1 to the litecoin.conf and it added the port itself, if i run TORLOG from the menus 'monitoring and configuration' (system) tor is working and passing data in and out.

and the system is not hanging during sync LTC, LND, TOR are all running.
i'll reboot it a few times after full sync and try to install RTL (that usually crashes on installing) and post back here
(i really loved the dietpi setup)

DrBlackross, 2021-09-30:
worked, can send and receive LTC ($10,etc), but as running a lightning node, no.
going to reinstall with raspiblitz as another btc node instead of ltc, and see if itll blend on amd86

DrBlackross, 2021-10-11:
ran both BTC and LTC installs (for chits) on stock ubuntu 18.04 amd64 with the build_sdcard script.... running into a macaroon issue

(i let it sit over the weekend while at a cat show)

*** Copy LND Macaroons to user admin ***
###### SYNCING MACAROONS, RPC Password AND TLS Certificate ######
.# make sure LND app-data directories exist

.# copy macaroons to central app-data directory and ensure unix ownerships and permissions
/bin/cp: cannot create regular file '/mnt/hdd/app-data/lnd/data/chain/bitcoin/mainnet/admin.macaroon': No such file or directory
/bin/cp: cannot create regular file '/mnt/hdd/app-data/lnd/data/chain/bitcoin/mainnet/invoice.macaroon': No such file or directory
/bin/cp: cannot create regular file '/mnt/hdd/app-data/lnd/data/chain/bitcoin/mainnet/readonly.macaroon': No such file or directory
/bin/cp: cannot create regular file '/mnt/hdd/app-data/lnd/data/chain/bitcoin/mainnet/invoices.macaroon': No such file or directory
/bin/cp: cannot create regular file '/mnt/hdd/app-data/lnd/data/chain/bitcoin/mainnet/chainnotifier.macaroon': No such file or directory
/bin/cp: cannot create regular file '/mnt/hdd/app-data/lnd/data/chain/bitcoin/mainnet/signer.macaroon': No such file or directory
/bin/cp: cannot create regular file '/mnt/hdd/app-data/lnd/data/chain/bitcoin/mainnet/walletkit.macaroon': No such file or directory
/bin/cp: cannot create regular file '/mnt/hdd/app-data/lnd/data/chain/bitcoin/mainnet/router.macaroon': No such file or directory
.# make sure admin has a symlink at ~/.lnd to /mnt/hdd/app-data/lnd/
.# make sure LND conf is readable and symlinked
.# make sure TLS certificate is readable and symlinked
OK - LND Macaroons created and copied

*** Check Wallet Lock ***
OK - Wallet is already unlocked

For debug logs CTRL+C and: tail -n1000 -f raspiblitz.log
or call the command 'debug' to see bigger report.
Starting the main menu ...
loading config data
WARN: LND needs re-setup
(i've re-setup enough, lol, same errors)

*** 70initLND.sh ***
*** Check Basic Config ***
*** Check bitcoin Running ***
*** Check bitcoin Responding *** (can take a while)
OK - chainnetwork is working

(i've fumbled though vagrant for a bit this week, going to see if i can it to work that way, lol)

DrBlackross, 2021-10-13:
[image]
it says 1.60, but it pulled 1.7.1 parts.... you'd have to check

admin@debian:~$ sudo blkid
/dev/sda1: UUID="48ee2bf7-0cd5-44c4-b062-002b709cd0c7" TYPE="ext4" PARTUUID="3970a32f-01"
/dev/sda5: UUID="98154c9c-fcf8-40ad-a75a-0f71733d05fa" TYPE="swap" PARTUUID="3970a32f-05"
/dev/sdb1: LABEL="BLOCKCHAIN" UUID="8d2096c0-3445-4241-a2ac-a8102ea3d62d" TYPE="ext4" PARTLABEL="primary" PARTUUID="11b52a4b-0abc-46fa-a157-8043f3c98514"

admin@debian:~$ free -h
              total        used        free      shared  buff/cache   available
Mem:          987Mi       866Mi        49Mi       0.0Ki        71Mi        15Mi
Swap:         2.9Gi       967Mi       2.0Gi
(i will add more ram LoL)

DrBlackross, 2021-11-16:
OK NEW ISSUE, been mucking with trying to ptach 1.6 up to 1.7.1
here's how i ran it up from the Vagrantfile setup above

vagrant up
ctrl+C at raspiblitz setup screen

sudo apt update 
yes to all changes

sudo apt upgrade -y
yes on pam setup screen <---- if you say no, you won't able to login to VM again after reboot
NO ON GRUB setup screen <---- grub will make the VM unbootable

sudo install usbutils -y <--- it needs this to patch 1.60 up to 1.7.1 from base vagrant install
reboot

vagrant ssh back in
menu
goto update, then patch, patch menu, select dev branch, select cancel after back to menu
goto main menu, update, then patch to 1.7.1, patch menu, patch
reboot (pray)

vagrant ssh 
fresh setup
wipe data
name node
(ill stick with LND)
New node
set passwords

│ ---> Blockchain Testrun              │  
│                                      │  
│ Run 'tail -f ./raspiblitz.log' in    │  
│ new terminal to follow install logs  |

system hangs here

admin@debian:~$ tail -f ./raspiblitz.log
hddGigaBytes=16
hddBlocksBitcoin=0
hddBlocksLitecoin=0
hddGotMigrationData=
-# migrationFile()
-# setting PASSWORD A

OK - password A changed for user pi, root, admin & bitcoin
...
Calling _provision.setup.sh for basic setup tasks ..

EOF

and i too am stuck

DrBlackross, 2021-11-16:
OH i take that back, it finally dropped a error log after an hour

###########################################################
-# /home/admin/raspiblitz.log

***********************************************
Running RaspiBlitz Bootstrap 1.7.1
Tue 16 Nov 2021 09:07:43 PM UTC
***********************************************
-# *** blitz.ssh.sh checkrepair
-# countKeyFiles(6)
-# DETECTED: MISSING SSHD KEYFILES --> Generating new ones
-# ssh-keygen1
ssh-keygen: generating new host keys: DSA 
-# countKeyFiles(8)
-# OK: New sshd host keys generated
-# OK: SSHD RUNNING
No HDMI switch found. 
No FS EXPAND needed. needsExpansion(0) fsexpanded(0)
No SSHRESET switch found. 
No UASP FIX needed (1st-try).
Waiting for HDD/SSD ...
total 600
drwxr-xr-x  2 root root   4096 Nov 16 21:07 .
drwxr-xr-x 85 root root   4096 Nov 16 21:07 ..
-rw-r--r--  1 root root 565189 Jan 31  2020 moduli
-rw-r--r--  1 root root   1580 Jan 31  2020 ssh_config
-rw-r--r--  1 root root   3250 Jan 31  2020 sshd_config
-rw-------  1 root root   1381 Nov 16 21:07 ssh_host_dsa_key
-rw-r--r--  1 root root    601 Nov 16 21:07 ssh_host_dsa_key.pub
-rw-------  1 root root    505 Oct 12  2020 ssh_host_ecdsa_key
-rw-r--r--  1 root root    173 Oct 12  2020 ssh_host_ecdsa_key.pub
-rw-------  1 root root    399 Oct 12  2020 ssh_host_ed25519_key
-rw-r--r--  1 root root     93 Oct 12  2020 ssh_host_ed25519_key.pub
-rw-------  1 root root   1823 Oct 12  2020 ssh_host_rsa_key
-rw-r--r--  1 root root    393 Oct 12  2020 ssh_host_rsa_key.pub
wifiIsSet=0
hddBackupLocationAvailable=0
hddRestoreConfigAvailable=0
memRestoreConfigAvailable=0
wifiRestore=0
wifiBackup=0
No UASP FIX needed (2nd-try).
isMounted: 0
Skipping UASP deactivation ... cmdlineExists(0) hddAdapterUSB() hddAdapterUSAP(0)
HDD already part of system: 0
Temp mounting (1) data drive (sda)
Temp mounting (1) result: 0
HDD is there but not AutoMounted yet - Waiting for user Setup/Update
-# df 
Filesystem     1K-blocks    Used Available Use% Mounted on
udev             2005580       0   2005580   0% /dev
tmpfs             404172    5408    398764   2% /run
/dev/sda1       15415240 4812000   9800476  33% /
tmpfs            2020852       0   2020852   0% /dev/shm
tmpfs               5120       0      5120   0% /run/lock
tmpfs            2020852       0   2020852   0% /sys/fs/cgroup
tmpfs              32768       4     32764   1% /var/cache/raspiblitz
-# lsblk -o NAME,FSTYPE,LABEL 
NAME   FSTYPE LABEL
sda           
├─sda1 ext4   
├─sda2        
└─sda5 swap   
sdb           
-# RASPIBLITZ DATA DRIVE Status

-# BASICS
isMounted=0
isBTRFS=0
[... 149 lines cut ...]
-# (33/50) checking if bitcoind is running ... 
-# (34/50) checking if bitcoind is running ... 
-# (35/50) checking if bitcoind is running ... 
-# (36/50) checking if bitcoind is running ... 
-# (37/50) checking if bitcoind is running ... 
-# (38/50) checking if bitcoind is running ... 
-# (39/50) checking if bitcoind is running ... 
-# (40/50) checking if bitcoind is running ... 
-# (41/50) checking if bitcoind is running ... 
-# (42/50) checking if bitcoind is running ... 
-# (43/50) checking if bitcoind is running ... 
-# (44/50) checking if bitcoind is running ... 
-# (45/50) checking if bitcoind is running ... 
-# (46/50) checking if bitcoind is running ... 
-# (47/50) checking if bitcoind is running ... 
-# (48/50) checking if bitcoind is running ... 
-# (49/50) checking if bitcoind is running ... 
-# (50/50) checking if bitcoind is running ... 
FAIL: setup: failed bitcoin
***********************************************************
ERROR - please report to development team
***********************************************************
state(error) message(_provision.setup.sh fail)
https://github.com/rootzoll/raspiblitz#support
command to shutdown --> off

DrBlackross, 2021-11-16:
ugh, i can run the stock Vagrantfile on my desktop just fine.... but running it remotely in the basement, it doesn't run 
Copy iso file /home/drblackross/.vagrant.d/tmp/VBoxGuestAdditions_6.1.26.iso into the box /tmp/VBoxGuestAdditions.iso
it will just run with the 1.6.0 rootzroll/raspiblitz

DrBlackross, 2021-11-16:
(on a lark)
copied the /home/drblackross/.vagrant.d/boxes/rootzoll-VAGRANTSLASH-raspiblitz from my desktop to the basement box
re-ran vagrant up lol

DrBlackross, 2021-11-16:
back to the lsusb after patching
/home/admin/config.scripts/blitz.datadrive.sh: line 383: lsusb: command not found

(walking away for another month lol)

rootzoll, 2021-11-25:
So this seems to colide with the `HDD Adpater UASP support` .. will make runing this part in next v1.7.2 release only if a RaspberryPi is detected.

DrBlackross, 2021-12-01:
(back in country for a few weeks.... lol)
Taking another stab at writing out to a 1tb VDI file 
(just because i can)
i can out of the pull from github run vagrant no problems BUT, lsusb (usbutils) and it doesn't see the sdd... even though i can cfdisk /dev/sdd and see it, weird, i can run menu and patch the system. reboot the vm but for some reason the vm loses the ssh key and i can't ssh in to 192.168.1.130 or use vagrant ssh.

hmmm....

ok, below is what i tried this time, i'll be back in a few more weeks to check this thread and see where i left off.

vagrant plugin install vagrant-persistent-storage

.# -*- mode: ruby -*-
.# vi: set ft=ruby :

Vagrant.configure('2') do |config|

  config.vm.box = 'rootzoll/raspiblitz'
.#  config.vm.network "private_network", type: "dhcp"
  config.vm.network "public_network", ip: "192.168.1.130"
.####################################################################
  config.ssh.username = 'admin'
  config.ssh.password = 'raspiblitz'
  config.ssh.insert_key = true
.####################################################################
  config.vm.disk :disk, size: '16GB', primary: true
.####################################################################

config.persistent_storage.enabled = true
config.persistent_storage.location = "~/blockchain.vdi"
config.persistent_storage.size = 5000
config.persistent_storage.mountname = 'sdd'
config.persistent_storage.filesystem = 'ext4'
.# config.persistent_storage.mountpoint = '/mnt/hdd'
.# config.persistent_storage.volgroupname = 'myvolgroup'

.####################################################################
  config.vm.provider 'virtualbox' do |vb, override|
    vb.memory = 2048
  end

  config.vm.provision 'shell',
                      privileged: false,
                      run: 'always',
                      path: 'alternative.platforms/amd64/packer/scripts/init_vagrant.sh'
end

rootzoll, 2021-12-09:
In `dev` branch and coming with v1.7.2 on the `blitz.datadrive.sh` script will no check if commans `lsusb` is availabe.

closed 2022-05-02 by openoms