raspiblitz/raspiblitz #2583: use parted with -a optimal also for ext4

rootzoll, 2021-09-30:


rootzoll, 2021-11-30:
Code changes added. Needs testing in starting with a fresh fat32 formatted SSD and a new build sd card.

rootzoll, 2022-02-17:
Its working so far for the v1.7.2 release ... but there is still a `The resulting partition is not properly aligned for best performance` warning ... moving the issue to v1.8.1 for a second look:

https://blog.hqcodeshop.fi/archives/273-GNU-Parted-Solving-the-dreaded-The-resulting-partition-is-not-properly-aligned-for-best-performance.html

psilonux, 2023-02-25:
Could you please change the follow line #https://github.com/rootzoll/raspiblitz/blob/b589b57215b57a0a1a97153d354a35716c4669fa/home.admin/config.scripts/blitz.datadrive.sh#L680` 

to `parted -s /dev/${hdd} mkpart primary ext4 0% 100% 1>&2` to fix the alignment when partitioning?

dipunm, 2023-03-26:
I am setting up a new pi and using a 2TB drive and found this warning.
Specifically:

```bash
Warning: The resulting partition is not properly aligned for best performance: 2048s % 65535s != 0s
```

I discovered 2 things:

1. I found the exact same blog article and using their script, I get this:
```bash
Using detected optimal_io_size alignment of 33553920 bytes in calc
Calculated alignment for /dev/sda (gpt) is: 65535s

Create partition with:
# parted /dev/sda mkpart [name] [type] 65535s 3907000094s
Verify partition alignment with:
# parted /dev/sda align-check optimal 1
```

2. in the `blitz.datadrive.sh status` function, I get an error about bad input.
```bash
    if [ "${hdd}" != "" ]; then
      hddBytes=$(fdisk -l /dev/$hdd | grep GiB | cut -d " " -f 5)
      hddGigaBytes=$(echo "scale=0; ${hddBytes}/1024/1024/1024" | bc -l)
    fi
```

I think the issue is that the first line in my output uses `TiB` and not `GiB`:
```bash
Disk /dev/sda: 1.82 TiB, 2000398934016 bytes, 3907029168 sectors
Disk model: SN570 2TB       
Units: sectors of 1 * 512 = 512 bytes
Sector size (logical/physical): 512 bytes / 512 bytes
I/O size (minimum/optimal): 512 bytes / 33553920 bytes
```

qlrd, 2023-05-09:
```
Warning: The resulting partition is not properly aligned for best performance: 2048s % 65535s != 0s
```

This is happening to me with recommended SanDisk SSD Plus 2.5":

tested SSDs: `1TB` and `2TB`
tested branches: `v1.8` and `v1.9`
tested hardware: `orangepi 5B`
tested OS: `armbian`

rootzoll, 2023-12-14:
Will be part of v1.11.0 next release candidate

closed 2023-12-14 by rootzoll