raspiblitz/raspiblitz #2584: [CL] Spark wallet v0.3.0rc for BOLT12 offers in a GUI

openoms, 2021-09-30:
With C-lightning installed offers are available from the CLI already: https://github.com/openoms/raspiblitz/blob/cln-FAQ/FAQ.cl.md#offers

Spark wallet gives a nice interface in it's latest release candidate: https://github.com/shesek/spark-wallet/releases/tag/v0.3.0rc
[image]

[image]

[image]

cli example to request a BOLT12 invoice for 21 sats:
```
lightning-cli fetchinvoice  lno1pgvkxun9v96x2epqwa5hg6pq2dcxzuntyptkzmrvv46pugq0clnzex2k5d9umw3snfyqnn8a0tw2ujutv3x2eqd5m3vdgw3y0y 21000
```

rootzoll, 2021-10-01:
So we will include this into v1.7.1? Is the PR ready to merge?

openoms, 2021-10-01:
It is an RC, but the only GUI implementing offers. I think it would be good to make it available. The PR is ready for final testing in the next raspiblitz rc.