raspiblitz/raspiblitz #2587: Reorder SSH Settings Menu

rootzoll, 2021-10-01:
For the future it seems better fit to move the CLBOSS from SETTINGS to SERVICES? Because its more of an additional app then a setting on CL.

Also in general it could also make more sense to move all the LND settings into the LND submenu and the CL settings into the CL submenu.

So that in SETTINGS you really just find the settings that are about the basic RaspiBlitz & Layer 1. All settings about installed on-top layers & apps shoudl be in their special menu section.

openoms, 2021-10-01:
CLBOSS made it to the settings because it is the equivalent (but better) of the LND autopilot,but I agree with improving the grouping for a future release.

rootzoll, 2021-11-30:
Pushing this as an UX improvement to v1.8.0

closed 2023-11-14 by rootzoll