raspiblitz/raspiblitz #2594: Keep losing connections

datvebi, 2021-10-03:
Had my node offline for a few months. Flashed the latest software (1.7.0) and now I need to download the blockchain from when it went offline but my node cannot keep any connections more than 1, sometimes 2. I've run bitcoin-cli addnode "<IP>" "onetry" a few dozen times with various node IPs (Tor and IPv4), and temporarily I can get the connections up to 6, but then within minutes it falls back to 1 and sometimes zero. It's downloading about .2% a day when previously I did the whole blockchain in 4 days. 

Debug logs: https://termbin.com/aynl

openoms, 2021-10-04:
Logs look with bitcoind syncing. I'd expect finished by now.
Otherwise did you try a restart?

datvebi, 2021-10-10:
Restarting was able to hold a few more connections than before, but it still did drop connections to zero often. Regardless, my node powered through and is synced now. Closing.

closed 2021-10-10 by datvebi