raspiblitz/raspiblitz #2597: Update to LND v0.13.3 due to CVE-2021-41592

openoms, 2021-10-04:
A vulnerability deacribed in:
https://lists.linuxfoundation.org/pipermail/lightning-dev/2021-October/003257.html

Can use the UPDATE menu to download the version, but LND v0.13.3 should be the part of RaspiBlitz v1.7.1 as default.

openoms, 2021-10-04:
The latest version of C-lightning is not known to.be vulnerable:

cdecker in https://t.me/lightningd/7713:
>Trying to dig up the details from the internal discussion, it looks to me like c-lightining always enforced the dust limit be below the reserve amount which means we were handling things correctly, but I'll let Rusty confirm that.

rootzoll, 2021-10-09:
RC3 released on dev branch for final testing.