raspiblitz/raspiblitz #2598: crontab under bos

AnimaI, 2021-10-04:
I have a small script for bos.
The script works perfectly under the user bos.

I would like to execute the script for bos via crontab. I have tried crontab -e under root, bitcoin and bos. Crontab does not execute the script for bos for any user.

I also have lnd-charge that I use under the bitcoin user. I also have a script for lnd-charge that I run as bitcoin user in crontab -e without any problems.

It seems to me that the crontab in the user bos is ineffective.

the crontab is:
`*/5 * * * /home/bos/start-telegram/check-process.sh`

Unfortunately I don't have a log to show or I can't find it.
How can I run a script intended for bos with crontab -e?

openoms, 2021-10-04:
Try:
`*/5 * * * sudo -u bos /home/bos/check-process.sh`

AnimaI, 2021-10-05:
Thank you @openoms
I have already tried this and tested it again this morning with user bos, bitcoin and root. The script is not executed.

If I execute the same command under the user bos, it works. So it's not the script. 
The permissions look like this: 
`-rwxr-xr-x  1 bos bos 1041 Oct  4 16:22 check-process.sh`

Where can I check what the crontab is doing?
Maybe I can find a note.

AnimaI, 2021-10-05:
Here is my plan for a better understanding of what exactly I have in mind.

I have activated the telegram service under BOS. With this command, bos telegram works perfectly:
`screen -dmS torify bos telegram --connect KEY-from-TelegramBot`

I put this command in a file with the name "telegram". 
In the file "telegram" there is only the command:
`torify bos telegram --connect myKEY-from-TelegramBot`
When I run the command ./telegram under the USER bos it works.

Basically, I just wanted a crontab that looks like this.
`@reboot bos sleep 120 && /home/bos/telegram >> /dev/null 2>&1`
I tried with all USER in the crontab -e, also with sudo -u. Nothing helped.

I thought it was the command @reboot or sleep that I applied in crontab.

So I wrote a small script that checks if the telegram script is running, and if not, it is started.

Here is the script:
```
#!/bin/bash

####################
#
# Script: Checking process and if it is not running will start that process
#
####################

#*** CONFIG ***

CMD_PATH="/home/bos/start-telegram/" # Root Path to Script
declare -a PROCESSES=('telegram'); # Script File Name

PRINT_TITLE_SCRIPT="Telegram on Start" # Script Title

#*** Vars ***
CHECK=$0

LOG_LOCATION=/dev/null
DATE=`date`

#*** Function ***
function monitor_process {
    local OUTPUT=$(ps ax | grep -v grep | grep -v $CHECK | grep $1)
    # Check the process if it is running?
    if [ "${#OUTPUT}" -gt 1 ] ; then
        # Server is running
        echo "  [*] Process ($1) is running."
    else
        # Service is not running, run the consumer process
        echo "  [x] Start Process ($1)"
        $CMD_PATH$1 > $LOG_LOCATION &
        # $CMD_PATH$1 &
    fi
}

#*** Start ***
echo " *** Running - Start $PRINT_TITLE_SCRIPT ( $DATE ) ***"

for var in "${PROCESSES[@]}"
do
  monitor_process ${var}
done

echo " *** Finished - $PRINT_TITLE_SCRIPT ***"
exit
```

**Here is the test for the script**

If the telegram service is not active!
I do the following command under  USER root

```
root@raspberrypi:~# sudo -u bos /home/bos/start-telegram/check-process.sh
 *** Running - Start Telegram on Start ( Tue  5 Oct 13:36:48 BST 2021 ) ***
  [x] Start Process (telegram)
 *** Finished - Telegram on Start ***
```
Normally I should receive a message on my Telegram that looks like this.
Connected to NODE ⚡️ NAME 111111
but there is no such message in my Telegram

Here: .... it runs... the script does nothing... that's ok.
```
bos@raspberrypi:~/start-telegram $ /home/bos/start-telegram/check-process.sh
 *** Running - Start Telegram on Start ( Tue  5 Oct 13:41:53 BST 2021 ) ***
  [*] Process (telegram) is running.
 *** Finished - Telegram on Start ***
```

I have found out that the script works under the USER bos.

In the **USER bos** with crontab -e the script is **not executed.**
`#*/5 * * * * /home/bos/start-telegram/check-process.sh  2>&1`

In **USER admin and root** with crontab -e the script is **not executed**
`#*/5 * * * * sudo -u bos /home/bos/start-telegram/check-process.sh  2>&1`

AnimaI, 2021-10-05:
After restarting the node, I get the following after the command
```
root@raspberrypi:~# sudo -u bos /home/bos/start-telegram/check-process.sh
 *** Running - Start Telegram on Start ( Tue  5 Oct 15:03:01 BST 2021 ) ***
  [x] Start Process (telegram)
**bos is not in the sudoers file.  This incident will be reported.**
 *** Finished - Telegram on Start ***
```

openoms, 2021-10-05:
the `bos` user is not in sudoers so don't have `sudo` in any of the scripts running with the `bos` user.

make sure lines are not commented out in the crontab and use absolute paths like:

`30 */4 * * * sudo -u bos /home/bos/.npm-global/bin/bos reconnect`

this applies to the scripts also.

AnimaI, 2021-10-05:
Thank`s Perfect, 
through your tip I found the solution.

I only changed the file "telegram" to
`screen -dmS torify /home/bos/.npm-global/bin/bos telegram --connect myKEY-from-TelegramBot`

now the crontab runs every 5 min and checks the script if the Telegram service of BOS is running.
`*/5 * * * * /home/bos/start-telegram/check-process.sh >> /home/bos/start-telegram/cronlog.log 2>&1`

It would suffice to create a crontab with
`@reboot bos sleep 300 && /home/bos/telegram >> /dev/null 2>&1`

The script "check-process.sh" checks whether the service has started and restarts if necessary. It's nice but almost excessive for such a small thing. We do everything to find a solution. :-)

openoms  thank you for taking the time to do this.

ChuckNorrison, 2022-09-25:
to get bos run as a service is the way to go. Service file will restart on bot crashes and keeps care on everything and logging will be much better.

This is my working service file

> [Unit]
> Description=Balance of Satoshis Telegram Bot
> Wants=network-online.target
> After=network-online.target
> 
> [Service]
> Type=simple
> WorkingDirectory=/home/bos/balanceofsatoshis
> ExecStart=/home/bos/.npm-global/bin/bos telegram --connect 123456789
> User=bos
> Group=bos
> Restart=always
> TimeoutSec=120
> RestartSec=15
> 
> [Install]
> WantedBy=multi-user.target

rootzoll, 2022-09-28:
@openoms should changing to systemd instead of crontab be a new issue?

openoms, 2022-09-28:
It could even be a PR or just noted in the FAQ

rootzoll, 2022-10-06:
@ChuckNorrison woudl you mind opening a new PR (or issue) on this?

ChuckNorrison, 2022-10-06:
> @ChuckNorrison woudl you mind opening a new PR (or issue) on this?

This looks more like a feature, the bos installation currently does not setup a cron or service. This feature would need a menu entry to setup bos telegram and create the desired service file. 

The process needs a telegram bot creation and api key from telegram. the bos will ask for this if not setup. This is a bit special and maybe not that easy to solve without starting bos manually. A documentation for the service file solution via menu entry could be easy.

ChuckNorrison, 2022-11-16:
made a PR with menu options to install telegram as service
https://github.com/rootzoll/raspiblitz/pull/3446

closed 2022-11-16 by rootzoll